wolfSSL/wolfssl #1568: Configure options do not carry over to installation

denravonska, 2018-05-18:
Maybe I'm using the library incorrectly but when I build wolfssl using configure and use that installation with wolfmqtt it seems like the options I passed to wolfssl's configure script do not follow into the installation.

For example, I build using `--enable-singlethreaded` yet wolfmqtt's configure does not have this information. If I point wolfmqtt to the wolfssl install dir using `--with-libwolfssl-prefix` it fails with

> wolfssl/wolfcrypt/wc_port.h:134: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'wolfSSL_Mutex'

which is due to SINGLE_THREADED not being defined.

dgarske, 2018-05-18:
Hi denravonska,

When building wolfSSL with `./configure` the generated defines are placed into `wolfssl/options.h`. You'll notice some include magic in wolfMQTT here: https://github.com/wolfSSL/wolfMQTT/blob/master/wolfmqtt/mqtt_socket.h#L36

If the define `WOLFSSL_USER_SETTINGS` is not set then we try and include options.h to get the wolfSSL build defines. This should allow you to synchronize the wolfSSL options used when building both libraries. When you do a 'sudo make install' it should take the options.h file as well. On my system its placed into the shared include path `/usr/local/include/wolfssl/options.h`.

Perhaps you can provide the exact build options you are using for wolfSSL and wolfMQTT? Its possible if you are cross-compiling the options.h used on the wolfMQTT make isn't getting populated. In which case I recommend using a shared user_settings.h file such as this template https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h and defining `WOLFSSL_USER_SETTINGS` in both ./configure steps (`./configure CFLAGS="-D WOLFSSL_USER_SETTINGS"`).

Thanks,
David Garske, wolfSSL

denravonska, 2018-05-18:
You are right, `wolfssl/options.h` does indeed contain `#define SINGLE_THREADED`.

wolfssl:
```
Vendor/wolfSSL/configure \
  --prefix=bin/avr32/release \
  --enable-nullcipher \
  --enable-singlethreaded \
  --enable-static \
  --disable-asyncthreads \
  --disable-shared \
  --disable-aes \
  --disable-aesgcm \
  --disable-filesystem \
  --disable-rsa \
  --disable-dh \
  --disable-md5 \
  --disable-sha \
  --disable-errorstrings \
  --disable-crypttests \
  --enable-smallstack \
  --enable-ecc \
  --enable-lowresource \
  --disable-examples \
  --disable-fips \
  --disable-selftest \
  --disable-harden \
  --enable-tlsv10 \
  --enable-fastmath \
  --host=avr32 --target=avr32 \
  CFLAGS=-Wpointer-arith -mrelax -mno-cond-exec-before-reload -Wstrict-prototypes -Werror-implicit-function-declaration -march=ucr3fp -mpart=uc3c2512c -mhard-float -std=gnu99 -ffunction-sections -fdata-sections -DWOLFSSL_STATIC_PSK -DNO_WRITEV -DWOLFSSL_USER_IO -DALT_ECC_SIZE
```

which properly defines SINGLE_THREADED:
```
grep SINGLE_THREADED bin/release/include/wolfssl/options.h 
#undef  SINGLE_THREADED
#define SINGLE_THREADED
```

wolfmqtt:

```
Vendor/wolfMQTT/configure \
  --prefix=bin/avr32/release \
  --with-libwolfssl-prefix=bin/avr32/release \
  --enable-tls \
  --enable-static \
  --disable-shared \
  --enable-nonblock \
  --disable-examples \
  --host=avr32 \
  --target=avr32 \
  CFLAGS="-Wpointer-arith -mrelax -mno-cond-exec-before-reload -Wstrict-prototypes -Werror-implicit-function-declaration -march=ucr3fp -mpart=uc3c2512c -mhard-float -std=gnu99 -ffunction-sections -fdata-sections"
```
The build line in the configure script executes is:

```
configure:14904: avr32-gcc -o conftest -Wpointer-arith -mrelax -mno-cond-exec-before-reload -Wstrict-prototypes -Werror-implicit-function-declaration -march=ucr3fp -mpart=uc3c2512c -mhard-float -std=gnu99 -ffunction-sections -fdata-sections  -fvisibility=hidden -Ibin/avr32/release/include -L/bin/avr32/release/lib -L/usr/local/lib conftest.c  -lwolfssl -lwolfssl >&5
```

If I put a `#error` in `options.h` it seems like it never gets included. It does use the correct install location according to the configure error log:

```
...
In file included from bin/avr32/release/include/wolfssl/ssl.h:33,
                 from conftest.c:28:
...
```

denravonska, 2018-05-18:
Program used in configure:

```
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "wolfmqtt"
| #define PACKAGE_TARNAME "wolfmqtt"
| #define PACKAGE_VERSION "1.0.0"
| #define PACKAGE_STRING "wolfmqtt 1.0.0"
| #define PACKAGE_BUGREPORT "https://github.com/wolfssl/wolfMQTT/issues"
| #define PACKAGE_URL "http://www.wolfssl.com"
| #define STDC_HEADERS 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_STRING_H 1
| #define HAVE_STRINGS_H 1
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_UNISTD_H 1
| #define LT_OBJDIR ".libs/"
| #define HAVE_VISIBILITY 1
| #define STDC_HEADERS 1
| #define SIZEOF_LONG_LONG 8
| #define SIZEOF_LONG 4
| #define HAVE_GETTIMEOFDAY 1
| #define HAVE_MEMSET 1
| #define HAVE_SIGNAL 1
| #define DEBUG 0
| /* end confdefs.h.  */
| 
|     #include <wolfssl/ssl.h>
| 
| int
| main ()
| {
| 
|     wolfSSL_Init();
| 
|   ;
|   return 0;
| }
configure:14914: result: no
```

eddy103-hub, 2021-02-24:
Hallo! Is there a documentation on how to use the library on Harmony V3?