wolfSSL/wolfssl #1576: QNX 6.6 cross compilation error

dagonzal01, 2018-05-22:
Hi,

I'm tring to crosscompile wolfssl library using Ubuntu as host OS, QNX 6.6 as target OS and with the next compilation options (./configure options):

`./configure CFLAGS=-Os CPPFLAGS=-DWOLFSSL_HAVE_MIN --host=arm-unknown-nto-qnx6.6.0eabi --enable-opensslextra --enable-fortress --enable-aesgcm --enable-ecc --enable-eccencrypt --enable-tlsx --enable-hkdf --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-sessioncerts --enable-cmac --enable-keygen --enable-testcert --enable-certgen`

However the compilation fails. I downloaded a copy of the master branch with [https://github.com/wolfSSL/wolfssl/commit/be9ae9a3c45ff692eac264ecf95fc50139a38f97](url) as latest commit.

Log:

damian@damian-VirtualBox:~/shared_with_windows/wolfssl-3.14.0$ ./configure CFLAGS=-Os CPPFLAGS=-DWOLFSSL_HAVE_MIN --host=arm-unknown-nto-qnx6.6.0eabi --enable-opensslextra --enable-fortress --enable-aesgcm --enable-ecc --enable-eccencrypt --enable-tlsx --enable-hkdf --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-sessioncerts --enable-cmac --enable-keygen --enable-testcert --enable-certgen
checking build system type... x86_64-unknown-linux-gnu
checking host system type... arm-unknown-nto-qnx6.6.0eabi
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for arm-unknown-nto-qnx6.6.0eabi-strip... arm-unknown-nto-qnx6.6.0eabi-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '1000' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking how to print strings... printf
checking for style of include used by make... GNU
checking for arm-unknown-nto-qnx6.6.0eabi-gcc... arm-unknown-nto-qnx6.6.0eabi-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-unknown-nto-qnx6.6.0eabi-gcc accepts -g... yes
checking for arm-unknown-nto-qnx6.6.0eabi-gcc option to accept ISO C89... none needed
checking whether arm-unknown-nto-qnx6.6.0eabi-gcc understands -c and -o together... yes
checking dependency style of arm-unknown-nto-qnx6.6.0eabi-gcc... gcc3
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by arm-unknown-nto-qnx6.6.0eabi-gcc... /home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-ld
checking if the linker (/home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-nm -B
checking the name lister (/home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to arm-unknown-nto-qnx6.6.0eabi format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-ld option to reload object files... -r
checking for arm-unknown-nto-qnx6.6.0eabi-objdump... arm-unknown-nto-qnx6.6.0eabi-objdump
checking how to recognize dependent libraries... pass_all
checking for arm-unknown-nto-qnx6.6.0eabi-dlltool... no
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for arm-unknown-nto-qnx6.6.0eabi-ar... arm-unknown-nto-qnx6.6.0eabi-ar
checking for archiver @FILE support... @
checking for arm-unknown-nto-qnx6.6.0eabi-strip... (cached) arm-unknown-nto-qnx6.6.0eabi-strip
checking for arm-unknown-nto-qnx6.6.0eabi-ranlib... arm-unknown-nto-qnx6.6.0eabi-ranlib
checking command to parse /home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-nm -B output from arm-unknown-nto-qnx6.6.0eabi-gcc object... ok
checking for sysroot... no
checking for arm-unknown-nto-qnx6.6.0eabi-mt... no
checking for mt... mt
configure: WARNING: using cross tools not prefixed with host triplet
checking if mt is a manifest tool... no
checking how to run the C preprocessor... arm-unknown-nto-qnx6.6.0eabi-gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
[... 316 lines cut ...]
   * Fast RSA:                   no
   * Single Precision:           no
   * Async Crypto:               no
   * Cavium:                     no
   * ARM ASM:                    no
   * AES Key Wrap:               no
   * Write duplicate:            no
   * Intel Quick Assist:         no
   * Xilinx Hardware Acc.:       no
   * Inline Code:                yes

---
damian@damian-VirtualBox:~/shared_with_windows/wolfssl-3.14.0$ make
make: Warning: File `aminclude.am' has modification time 474 s in the future
make -j2  all-am
make[1]: Entering directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[1]: Warning: File `aminclude.am' has modification time 470 s in the future
  CC       wolfcrypt/src/src_libwolfssl_la-integer.lo
In file included from wolfcrypt/src/integer.c:41:0:
./wolfcrypt/src/misc.c:252:26: error: expected ')' before 'a'
./wolfcrypt/src/misc.c:252:26: error: expected ')' before '>' token
./wolfcrypt/src/misc.c:252:26: error: expected ')' before '?' token
make[1]: *** [wolfcrypt/src/src_libwolfssl_la-integer.lo] Error 1
make[1]: Leaving directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make: *** [all] Error 2

kaleb-himes, 2018-05-22:
Hi @dagonzal01,

It looks like you're specifying an alternate host with ```--host=arm-unknown-nto-qnx6.6.0eabi``` but your compiler isn't being specified for cross compilation, this would typically be done something like:

```
./configure --host=arm-unknown-nto-qnx6.6.0eabi \
CC=/path/to/toolchain/bin/arm-unknown-nto-qnx6.6.0eabi-gcc \
AR=/path/to/toolchain/bin/arm-unknown-nto-qnx6.6.0eabi-ar \
AS=/path/to/toolchain/bin/arm-unknown-nto-qnx6.6.0eabi-gcc \
RANLIB=/path/to/toolchain/bin/arm-unknown-nto-qnx6.6.0eabi-ranlib \
LD=/path/to/toolchain/bin/arm-unknown-nto-qnx6.6.0eabi-ld \
<now CFLAG options and other configure options here>
```

Could you try that and let us know your results?

Warm Regards,

Kaleb

kaleb-himes, 2018-05-23:
Hi @dagonzal01,

Disregard my last, I see you have the toolchain in your path and it is being properly detected as we see in the configure log example:

```
checking for ld used by arm-unknown-nto-qnx6.6.0eabi-gcc... /home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-ld
checking if the linker (/home/damian/qnx660/host/linux/x86/usr/bin/arm-unknown-nto-qnx6.6.0eabi-ld) is GNU ld... yes
```

I suspect then you must be getting a conflict somewhere. Could you try using these CFLAGS instead:

CFLAGS="-Os -DWOLFSSL_HAVE_MIN -DWOLFSSL_HAVE_MAX" and let me know if adding the have max resolves the conflict.

Warm Regards,

Kaleb

dagonzal01, 2018-05-23:
Hi @kaleb-himes,

Thanks for your answer. I modified a little bit your suggestion like this:

`./configure CFLAGS=-Os CPPFLAGS="-DWOLFSSL_HAVE_MIN -DWOLFSSL_HAVE_MAX" --host=arm-unknown-nto-qnx6.6.0eabi --enable-opensslextra --enable-fortress --enable-aesgcm --enable-ecc --enable-eccencrypt --enable-tlsx --enable-hkdf --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-sessioncerts --enable-cmac --enable-keygen --enable-testcert`

And the build process was successful :)

However, "make check" was not. This was the output of the command and test-suite.log:

#####make check#####

damian@damian-VirtualBox:~/shared_with_windows/wolfssl-3.14.0$ make check
make -j2  check-am
make[1]: Entering directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make -j2   testsuite/testsuite.test tests/unit.test   scripts/resume.test  scripts/ocsp.test scripts/ocsp-stapling.test scripts/ocsp-stapling-with-ca-as-responder.test scripts/ocsp-stapling2.test     scripts/external.test scripts/google.test scripts/openssl.test
make[2]: Entering directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[2]: warning: -jN forced in submake: disabling jobserver mode.
make[2]: Nothing to be done for `scripts/resume.test'.
make[2]: Nothing to be done for `scripts/ocsp.test'.
make[2]: Nothing to be done for `scripts/ocsp-stapling.test'.
make[2]: Nothing to be done for `scripts/ocsp-stapling-with-ca-as-responder.test'.
make[2]: Nothing to be done for `scripts/ocsp-stapling2.test'.
make[2]: Nothing to be done for `scripts/external.test'.
make[2]: Nothing to be done for `scripts/google.test'.
make[2]: Nothing to be done for `scripts/openssl.test'.
make[2]: `tests/unit.test' is up to date.
make[2]: Leaving directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make -j2  check-TESTS
make[2]: Entering directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[2]: warning: -jN forced in submake: disabling jobserver mode.
make[3]: Entering directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[3]: warning: -jN forced in submake: disabling jobserver mode.
PASS: scripts/resume.test
PASS: scripts/external.test
PASS: scripts/ocsp.test
PASS: scripts/openssl.test
PASS: scripts/google.test
FAIL: scripts/ocsp-stapling.test
FAIL: testsuite/testsuite.test
FAIL: scripts/ocsp-stapling-with-ca-as-responder.test
FAIL: tests/unit.test
PASS: scripts/ocsp-stapling2.test
make[4]: Entering directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[4]: warning: -jN forced in submake: disabling jobserver mode.
make -j2  all-am
make[5]: Entering directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[5]: warning: -jN forced in submake: disabling jobserver mode.
make[5]: Leaving directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[4]: Leaving directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
============================================================================
Testsuite summary for wolfssl 3.14.0
============================================================================
# TOTAL: 10
# PASS:  6
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
make[3]: *** [test-suite.log] Error 1
make[3]: Leaving directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[2]: *** [check-TESTS] Error 2
make[2]: Leaving directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make[1]: *** [check-am] Error 2
make[1]: Leaving directory `/home/damian/shared_with_windows/wolfssl-3.14.0'
make: *** [check] Error 2

#####test-suite.log#####

======================================
   wolfssl 3.14.0: ./test-suite.log
======================================

# TOTAL: 10
# PASS:  6
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: testsuite/testsuite
=========================

./testsuite/testsuite.test: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/testsuite/.libs/lt-testsuite.test: cannot execute binary file: Exec format error
./testsuite/testsuite.test: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/testsuite/.libs/lt-testsuite.test: Success

FAIL: tests/unit
================

./tests/unit.test: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/tests/.libs/lt-unit.test: cannot execute binary file: Exec format error
./tests/unit.test: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/tests/.libs/lt-unit.test: Success

FAIL: scripts/ocsp-stapling
===========================

./examples/client/client: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/examples/client/.libs/lt-client: cannot execute binary file: Exec format error
./examples/client/client: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/examples/client/.libs/lt-client: Success
-e 

Client connection failed

FAIL: scripts/ocsp-stapling-with-ca-as-responder
================================================

./examples/client/client: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/examples/client/.libs/lt-client: cannot execute binary file: Exec format error
./examples/client/client: line 117: /home/damian/shared_with_windows/wolfssl-3.14.0/examples/client/.libs/lt-client: Success
-e 

Client connection failed

Best regards,

Damian

kaleb-himes, 2018-05-23:
Hi @dagonzal01,

Just to be sure, the make check would only succeed if run on the target architecture. We would not expect this to pass for example if you compiled for a host arm-unknown-nto-qnx6.6.0eabi but then ran the make check on a non-qnx architecture.

To get the make check passing in this scenario you would want to add ```--disable-examples``` to the build settings.

Warm Regards,

KH

dagonzal01, 2018-05-23:
Hi @kaleb-himes,

Thank you for your quick answer. After adding `--disable-examples`, `make check` succeeded.

Best regards,

Damian

closed 2018-05-23 by dagonzal01