wolfSSL/wolfssl #1615: Trouble compiling w/ gcc8 on a fresh Arch Linux install

maximevince, 2018-06-11:
Hi guys,

Not sure if this is me, or just because of the lastest gcc update. (gcc version 8.1.1 on Arch Linux)
But after a simple:
$ ./autogen.sh
$ ./configure
$ make

I am getting these build errors:

```
  CC       wolfcrypt/src/src_libwolfssl_la-poly1305.lo
In file included from wolfcrypt/src/wc_port.c:28:
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirFirst’:
./wolfssl/wolfcrypt/types.h:321:32: error: ‘strncat’ specified bound 1 equals source length [-Werror=stringop-overflow=]
      #define XSTRNCAT(s1,s2,n) strncat((s1),(s2),(n))
                                ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:269:9: note: in expansion of macro ‘XSTRNCAT’
         XSTRNCAT(ctx->name, "/", 1);
         ^~~~~~~~
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirNext’:
./wolfssl/wolfcrypt/types.h:321:32: error: ‘strncat’ specified bound 1 equals source length [-Werror=stringop-overflow=]
      #define XSTRNCAT(s1,s2,n) strncat((s1),(s2),(n))
                                ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:316:9: note: in expansion of macro ‘XSTRNCAT’
         XSTRNCAT(ctx->name, "/", 1);
         ^~~~~~~~
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirFirst’:
./wolfssl/wolfcrypt/types.h:321:32: error: ‘strncat’ output may be truncated copying 128 bytes from a string of length 255 [-Werror=stringop-truncation]
      #define XSTRNCAT(s1,s2,n) strncat((s1),(s2),(n))
                                ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:270:9: note: in expansion of macro ‘XSTRNCAT’
         XSTRNCAT(ctx->name, ctx->entry->d_name, MAX_FILENAME_SZ/2);
         ^~~~~~~~
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirNext’:
./wolfssl/wolfcrypt/types.h:321:32: error: ‘strncat’ output may be truncated copying 128 bytes from a string of length 255 [-Werror=stringop-truncation]
      #define XSTRNCAT(s1,s2,n) strncat((s1),(s2),(n))
                                ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:317:9: note: in expansion of macro ‘XSTRNCAT’
         XSTRNCAT(ctx->name, ctx->entry->d_name, MAX_FILENAME_SZ/2);
         ^~~~~~~~

```

I suspect this has to do with new default warning/error configuration of gcc 8.

maximevince, 2018-06-11:
Probably related to:
https://github.com/gcc-mirror/gcc/commit/d8aad7864e5b4c654dcea86b98085baf36d8db76

ejohnstown, 2018-06-11:
I think it makes sense. The single byte strncat() won't null terminate the dst string. (Luckily the dst is zeroed out.) The other error is similar. The ctx->name string build should probably be a little smarter.

ejohnstown, 2018-06-12:
As a workaround for now, you could add the following to the configure CFLAGS on the command line:

    $ ./configure CFLAGS="-Wno-stringop-truncation -Wno-stringop-overflow -Wno-size-of-pointer-memaccess"
    $ make

This should build.

ejohnstown, 2018-06-25:
Pull request #1634 fixes the indicated issues, and I tracked down several more. Thanks for the issue report.

closed 2018-06-25 by ejohnstown