wolfSSL/wolfssl #1661: Problem with the Python wrapper

mikemeier999, 2018-07-05:
Hi,

I get an Error if want to "make check" the python wrapper for wolfssl. I took the tutorial
https://github.com/wolfSSL/wolfssl/tree/master/wrapper/python/wolfssl/README.rst. The wolfssl C and requirements-testing.txt installation works fine, no mistake.  But if I run "make check" there are Errors with py27 and py36.

The summary of the "make check":
_______________________________________________ summary _______________________________________________
ERROR:   py27: InvocationError for command /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/py27/bin/pip install -U --no-deps /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/dist/wolfssl-0.1.0.zip (see /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/py27/log/py27-10.log) (exited with code 1)
SKIPPED:  py34: InterpreterNotFound: python3.4
SKIPPED:  py35: InterpreterNotFound: python3.5
ERROR:   py36: InvocationError for command /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/py36/bin/pip install -U --no-deps /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/dist/wolfssl-0.1.0.zip (see /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/py36/log/py36-10.log) (exited with code 1)
Makefile:54: recipe for target 'test' failed
make: *** [test] Error 1

And if I want to install "/home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/py27/bin/pip install -U --no-deps /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/dist/wolfssl-0.1.0.zip" manual, there is the following output:

jens@desktop-1804:~/Programme/wolfssl/wrapper/python/wolfssl/.tox/py27/bin$ pip install -U --no-deps /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/dist/wolfssl-0.1.0.zip 
Processing /home/jens/Programme/wolfssl/wrapper/python/wolfssl/.tox/dist/wolfssl-0.1.0.zip
Building wheels for collected packages: wolfssl
  Running setup.py bdist_wheel for wolfssl ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-qZvLhb-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp14CJ98pip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-2.7
  creating build/lib.linux-x86_64-2.7/wolfssl
  copying src/wolfssl/_memory.py -> build/lib.linux-x86_64-2.7/wolfssl
  copying src/wolfssl/__about__.py -> build/lib.linux-x86_64-2.7/wolfssl
  copying src/wolfssl/_methods.py -> build/lib.linux-x86_64-2.7/wolfssl
  copying src/wolfssl/exceptions.py -> build/lib.linux-x86_64-2.7/wolfssl
  copying src/wolfssl/__init__.py -> build/lib.linux-x86_64-2.7/wolfssl
  copying src/wolfssl/utils.py -> build/lib.linux-x86_64-2.7/wolfssl
  copying src/wolfssl/build_ffi.py -> build/lib.linux-x86_64-2.7/wolfssl
  running build_ext
  generating cffi module 'build/temp.linux-x86_64-2.7/wolfssl._ffi.c'
  creating build/temp.linux-x86_64-2.7
  building 'wolfssl._ffi' extension
  creating build/temp.linux-x86_64-2.7/build
  creating build/temp.linux-x86_64-2.7/build/temp.linux-x86_64-2.7
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-nbjU53/python2.7-2.7.15~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/local/include -I/usr/include/python2.7 -c build/temp.linux-x86_64-2.7/wolfssl._ffi.c -o build/temp.linux-x86_64-2.7/build/temp.linux-x86_64-2.7/wolfssl._ffi.o
  build/temp.linux-x86_64-2.7/wolfssl._ffi.c:496:10: error: conflicting types for ‘wolfSSL_Free’
       void wolfSSL_Free(void *ptr, void* heap, int type);
            ^~~~~~~~~~~~
  In file included from /usr/local/include/wolfssl/wolfcrypt/types.h:244:0,
                   from /usr/local/include/wolfssl/wolfcrypt/logging.h:33,
                   from /usr/local/include/wolfssl/ssl.h:35,
                   from build/temp.linux-x86_64-2.7/wolfssl._ffi.c:494:
  /usr/local/include/wolfssl/wolfcrypt/memory.h:71:27: note: previous declaration of ‘wolfSSL_Free’ was here
           WOLFSSL_API void  wolfSSL_Free(void *ptr);
                             ^~~~~~~~~~~~
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for wolfssl
  Running setup.py clean for wolfssl
Failed to build wolfssl
Installing collected packages: wolfssl
  Running setup.py install for wolfssl ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-qZvLhb-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-tCnr5v-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-2.7
    creating build/lib.linux-x86_64-2.7/wolfssl
    copying src/wolfssl/_memory.py -> build/lib.linux-x86_64-2.7/wolfssl
    copying src/wolfssl/__about__.py -> build/lib.linux-x86_64-2.7/wolfssl
    copying src/wolfssl/_methods.py -> build/lib.linux-x86_64-2.7/wolfssl
    copying src/wolfssl/exceptions.py -> build/lib.linux-x86_64-2.7/wolfssl
    copying src/wolfssl/__init__.py -> build/lib.linux-x86_64-2.7/wolfssl
    copying src/wolfssl/utils.py -> build/lib.linux-x86_64-2.7/wolfssl
    copying src/wolfssl/build_ffi.py -> build/lib.linux-x86_64-2.7/wolfssl
    running build_ext
    generating cffi module 'build/temp.linux-x86_64-2.7/wolfssl._ffi.c'
    creating build/temp.linux-x86_64-2.7
    building 'wolfssl._ffi' extension
    creating build/temp.linux-x86_64-2.7/build
    creating build/temp.linux-x86_64-2.7/build/temp.linux-x86_64-2.7
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-nbjU53/python2.7-2.7.15~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/local/include -I/usr/include/python2.7 -c build/temp.linux-x86_64-2.7/wolfssl._ffi.c -o build/temp.linux-x86_64-2.7/build/temp.linux-x86_64-2.7/wolfssl._ffi.o
    build/temp.linux-x86_64-2.7/wolfssl._ffi.c:496:10: error: conflicting types for ‘wolfSSL_Free’
         void wolfSSL_Free(void *ptr, void* heap, int type);
              ^~~~~~~~~~~~
    In file included from /usr/local/include/wolfssl/wolfcrypt/types.h:244:0,
                     from /usr/local/include/wolfssl/wolfcrypt/logging.h:33,
                     from /usr/local/include/wolfssl/ssl.h:35,
                     from build/temp.linux-x86_64-2.7/wolfssl._ffi.c:494:
    /usr/local/include/wolfssl/wolfcrypt/memory.h:71:27: note: previous declaration of ‘wolfSSL_Free’ was here
             WOLFSSL_API void  wolfSSL_Free(void *ptr);
                               ^~~~~~~~~~~~
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-qZvLhb-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-tCnr5v-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-qZvLhb-build/

Does anyone know what I have to do?

Regards

ejohnstown, 2018-07-06:
Thank you for the bug report. I'm going to look into what's going on.

ejohnstown, 2018-07-18:
I have a pull request open, #1701, that fixes this.

ejohnstown, 2018-07-18:
We have a more up to date version of our Python wrappers in the [wolfssl-py](https://github.com/wolfssl/wolfssl-py) repository. This issue was fixed over there.

closed 2018-07-18 by ejohnstown