wolfSSL/wolfssl #1746: Native TFO support

vlolteanu, 2018-08-08:
It would be nice if WolfSSL had proper TCP Fast Open support. This would basically mean two things:
 * adding a function call that registers the server's address with the WOLFSSL object and
 * replacing the first send() with sendto(MSG_FASTOPEN) or connectx() (or whatever other function is used to send a TFO SYN)

I have a slightly hackish Linux-centric implementation that does just that: you can replace connect() with wolfSSL_SetTFOAddr() and the library takes care of the rest. You can take a look at it here: https://github.com/vlolteanu/wolfssl/tree/linuxcentric

kaleb-himes, 2018-08-16:
@vlolteanu 

Thank you so much for the suggestion! I have added a feature request to our internal tracker:

```
Feature Request: Support for TCP Fast Open
Requestor: Vladimir Olteanu
Contact: @vlolteanu via github
Timeline: None specified at this time, can be accelerated via agreement at any time
Source of Request: https://github.com/wolfSSL/wolfssl/issues/1746
```

Given you have done some work on this already if you ever decide you would like to contribute that solution please shoot us an email at support@wolfssl.com and note you are interested in receiving a wolfSSL contributor agreement in order to make a contribution.

Warm Regards,

KH

muelli, 2018-09-02:
> replacing the first send() with sendto(MSG_FASTOPEN) or connectx() (or whatever other function is used to send a TFO SYN)

Since Linux 4.11 you can use `FASTOPEN_CONNECT` without the need to change you `send` and `recv` calls.

muelli, 2018-09-22:
Do you happen to have modified client and server examples which make use of TCPFO?

vlolteanu, 2018-09-22:
Nope, but I have used the modified API in one of my personal projects.

muelli, 2018-10-01:
should the following work with your new API?

```
diff --git a/examples/client/client.c b/examples/client/client.c
index 391d6fce..2ec6c469 100644
--- a/examples/client/client.c
+++ b/examples/client/client.c
@@ -1942,6 +1942,15 @@ THREAD_RETURN WOLFSSL_THREAD client_test(void* args)
         err_sys("unable to get SSL object");
     }
 
+    {
+        SOCKADDR_IN_T addr;
+
+        build_addr(&addr,  host, port, 0, 0);
+        wolfSSL_SetTFOAddr (ssl, (struct sockaddr_storage *) &addr, sizeof (addr));
+    }
+
     #ifdef OPENSSL_EXTRA
     wolfSSL_KeepArrays(ssl);
     #endif
```

My TCPFO packet has 8 bytes of payload that I can't explain. I expected to see a much bigger packet, anyway.

muelli, 2018-10-08:
FTR: I've added the sockopts for TCP Fast Open here: https://github.com/muelli/wolfssl/tree/tcpfo
There are also a few patches for printf debugging on that branch.

haydenroche5, 2021-01-13:
Please reach out if you're interested in us implementing this feature in a future release of wolfSSL. As Kaleb previously mentioned, we are tracking this request internally.

Thanks!

Hayden

closed 2021-01-13 by haydenroche5