wolfSSL/wolfssl #1761: --enable-distro vs. --enable-all

lechner, 2018-08-16:
Hi,

wolfSSL's Debian package uses `--enable-distro` to select features. That option was conceived to ensure a consistent set of features across Linux distributions. Upon request, you also introduced [shared object versioning](https://en.wikipedia.org/wiki/Soname), which dynamic linkers use when loading applications. People in Debian are [starting to use TLS 1.3](https://bugs.debian.org/904710), but it is not included in `--enable-distro`. (For now, it is enabled manually.) Would it be better for Debian to use `--enable-all`? Thank you!

Best regards,
Felix

dgarske, 2018-08-16:
Hi Felix! Great to hear from you! The —enable-distro option enables the —enable-all option already. I will discuss with the team the idea of adding tls13 to the —enable-all option.

dgarske, 2018-08-16:
Hi Felix,

I double checked and both `./configure --enable-distro` and `./configure --enable-all` enable TLS v1.3. This was added into master on May 5, 2017 in PR #877 and has been in all wolfSSL releases since v3.12.0-stable.

Let me know if you can confirm or reproduce an issue and we are happy to investigate.

Thanks,
David Garske, wolfSSL

lechner, 2018-08-22:
Closing, as I have not heard back from the Debian bug filer. Thank you!

closed 2018-08-22 by lechner