wolfSSL/wolfssl #1774: Makefiles for Python wrapper

lechner, 2018-08-22:
Hi, with some interest in Debian, I would like to package more features. The Python module in `wrapper/python/wolfssl/` comes with a Makefile, but I could not find one for `wrapper/python/wolfcrypt/`. Was there ever a second Makefile that was lost somehow? Thank you!

ejohnstown, 2018-09-07:
Felix:

There is another makefile, but it is in our wolfcrypt-py repository. The engineer who was working on it part time has moved onto other things and no one else has picked it up yet.

--John

ejohnstown, 2019-02-22:
Felix:

We have moved our wolfSSL and wolfCrypt wrappers for Python to their own projects.

1. https://github.com/wolfssl/wolfssl-py
2. https://github.com/wolfssl/wolfcrypt-py

At this point, I wouldn't use the Python builds in the source archive.

ejohnstown, 2019-02-26:
Also, which features were you looking to add? We recently expanded our API in the Python wrappers.

dgarske, 2019-03-14:
Hi Felix,

As always, thank you for your contributions to this project. Following up on this issue to make sure its fully resolved.

Do you have any feedback for John on the previous question about features to add?

Thanks,
David Garske, wolfSSL

ejohnstown, 2019-04-04:
Felix:

I'm going to close this issue at this time. If you want to make some suggestions for additions to the wolfSSL Python wrappers, please see the GitHub repository for [wolfSSL-py](https://github.com/wolfssl/wolfssl-py). Thank you.

--John

closed 2019-04-04 by ejohnstown