wolfSSL/wolfssl #1793: Issues using the ATECC508A/ATECC608A

WolfWalter, 2018-08-27:
Hi, 
I'm trying to use wolfssl in combination with wolfMqtt on a PIC32MZ2064DAH and the ATECC508A/ATECC608A. I had multiple issues but finally everything compiles and runs, but I get an Alert ("bad record mac") in the wolfssl debug output and Wireshark log. The same code without "WOLFSSL_ATECC508A" works as expected.
I had to add 
```
 #ifdef WOLFSSL_ATECC508A
    /* populate key->pubkey_raw */
    XMEMCPY(key->pubkey_raw, (byte*)in, sizeof(key->pubkey_raw));
#endif
```
after 
```
/* adjust to skip first byte */
    inLen -= 1;
    in += 1;
```
 in wc_ecc_import_x963_ex() (ecc.c). Otherwise the public_raw buffer for EccSharedSecret() was empty.

If I add the callbacks as suggested in the atmel [README.md](https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/port/atmel/README.md) the TLS-connection fails with BAD_COND_E. The corresponding code is commented with: 
> /* TODO: Implement equiv call to ATECC508A */

A [forums post](https://www.wolfssl.com/forums/topic965-integration-with-atmel-atecc508a.html) from 2017 suggest that the implementation is not finished?

Should this use-case be possible with wolfssl? I tried 3.15.3 and the current master version.

WolfWalter, 2018-08-29:
Hi dgarske,
thank you for the quick response. I already tried to define WOLFSSL_ATECC508A and HAVE_PK_CALLBACKS. But in atcatls_create_pms_cb() wc_ecc_import_unsigned() (in ecc.c) is called wich is a wrapper for wc_ecc_import_raw_private() (in ecc.c). This function returns BAD_COND_E because of:
```
#ifdef WOLFSSL_ATECC508A
    /* TODO: Implement equiv call to ATECC508A */
    err = BAD_COND_E;
    (void)d;
    (void)encType;

#else
...
```
What am I doing wrong here?

dgarske, 2018-08-29:
Hi WolfWalter,

To get that scenario to work do not define `WOLFSSL_ATECC508A`. Then you'll need to handle calling the `atcatls_init` as is done in `wolfcrypt/src/port/atmel/atmel.c:226' in function `atmel_init` yourself outside of wolfSSL.

Thanks,
David Garske, wolfSSL

WolfWalter, 2018-08-30:
Hi,
the issue persists with WOLFSSL_ATECC508A not defined. Still got the alert. I will try different servers next.

WolfWalter, 2018-08-31:
Hi,
I tried the wolfMQTT awsiot example and after solving some problems (not related to wolfssl) it works and uses the ATECC508A (ATECC608A works as well). I'm using the WOLFSSL_ATECC508A variant with the "pubkey_raw"-fix [(PR 1804).](https://github.com/wolfSSL/wolfssl/pull/1804)
I'm still not sure what the issue with the iot.eclispe.org-MQTT server is. Any ideas?

dgarske, 2018-09-06:
Hi WolfWalter,

Thanks for your fixes in PR #1804. Have you tried the fix in wolfMQTT (https://github.com/wolfSSL/wolfMQTT/pull/90 that resolves a new issue with packets over 128 bytes in size? That started with the MQTT v5 changes in https://github.com/wolfSSL/wolfMQTT/pull/87

Thanks,
David Garske, wolfSSL

WolfWalter, 2018-09-11:
Hi,
the PR-Version works for the aws-iot example but did not help with the issue at the mqttclient-example and the iot.eclipse.org server.

The current wolfMQTT master did not run on my board. I got a system-exception and was not yet able to track it down.

dgarske, 2018-09-11:
Hi WolfWalter,

It would be helpful to get a Wireshark trace and/or wolfSSL logs. Logging can be enabled by defining `DEBUG_WOLFSSL` and calling `wolfSSL_Debugging_ON()`. Can you also send us your build options (either your ./configure statement or user_settings.h)? If you'd like to start a support ticket just email us at support@wolfssl.com and reference this GitHub issue.

With iot.eclipse.org it requires RSA enabled. Looks like its using the cipher suite (TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384). Make sure NO_RSA is not defined. If using fast math (USE_FAST_MATH) Make sure FP_MAX_BITS is either not defined or set to maxkeysz * 2 = 4096.

Thanks,
David Garske, wolfSSL

WolfWalter, 2018-09-13:
Hi,
thank you for the support! Here are the requested files for the mqttclient-example (iot.eclipse.org):
[config.h.txt](https://github.com/wolfSSL/wolfssl/files/2378085/config.h.txt)
[console_log.txt](https://github.com/wolfSSL/wolfssl/files/2378086/console_log.txt)
[user_settings.h.txt](https://github.com/wolfSSL/wolfssl/files/2378087/user_settings.h.txt)
[wireshark_log.zip](https://github.com/wolfSSL/wolfssl/files/2378088/wireshark_log.zip)

`NO_RSA` is not defined and `FP_MAX_BITS` is not set.

dgarske, 2018-09-13:
Hi WolfWalter,

In your `console_log.txt`:

```
received record layer msg                                                       
got ALERT!                                                                      
Got alert                                                                       
wolfSSL error occurred, error = 20   
```

The alert code 20 is `bad_record_mac`.

That error along with the PIC32MZ is one I recognize and has been fixed recently. There were a few bugs with the PIC32MZ hashing and crypto hardware acceleration that caused a bad record mac, which have been fixed.

Those fixes are:
* https://github.com/wolfSSL/wolfssl/pull/1479 (in v3.15.3)
* https://github.com/wolfSSL/wolfssl/pull/1714 (GitHub master, not released yet)
* https://github.com/wolfSSL/wolfssl/pull/1779 (GitHub master, not released yet)

Will you confirm you are using a version of wolfSSL that contains these fixes?

Thanks,
David Garske, wolfSSL

WolfWalter, 2018-09-14:
Hi,
I started with v3.15.3 but I have also tried current versions of wolfssl all with the same result (bad record mac). For the logs I used your branch [d0c728b](https://github.com/dgarske/wolfssl/commit/d0c728b114d608c04be99e0688f61b16f1d7aaab).

I solved my issue using the current master of wolfMQTT but still got the same error.

dgarske, 2018-09-19:
Hi WolfWalter,

Can you try disabling the PIC32MZ crypto hardware to verify its not related? You'll need to define the following in your config.h or user_settings.h file.

```
#define NO_PIC32MZ_CRYPT
#define NO_PIC32MZ_HASH
```

If this resolves the issue then we know its related to the PIC32MZ crypto hardware. Their have been a handful of fixes for PIC32MZ going back father, most of which are in the latest Harmony. Which Harmony version are you using? Also Harmony uses us for the crypto engine and TLS, so there is duplicated code. They are trying to clean that up, but you may end up having to update wolfCrypt sources in `framework/crypto/src` and `third_party/tcpip/wolfssl`.

The `framework/crypto/src` is a combination of the wolfcrypt/src/*.c and wolfssl/wolfcrypt/*.h including the pic32mz-crypt.c and pic32mz-crypt.h files from wolfcrypt/src/port/pic32 and wolfssl/wolfcrypt/port/pic32/.

The `third_party/tcpip/wolfssl` is basically a straight copy of the last release.

Let us know the results.

Thanks,
David Garske, wolfSSL

WolfWalter, 2018-09-28:
Hi,
I already had deactivated the PIC32MZ crypto hardware and used the current wolfssl sources. Harmony-Version is 2.06. For now I'm out of ideas. Hopefully you find the time to integrate the ATECC-chips officially soon.

dgarske, 2018-10-16:
Hi @WolfWalter,

We discovered a bug in the ECC shared secret reference callback for the client side case that was not loading the peer key. It would be great if you could test again with the latest in PR #1815 (
https://github.com/wolfSSL/wolfssl/pull/1815/commits/e53694b351f49c93bc8aa4f8b11f2e22f9419bff). 

Thanks,
David Garske, wolfSSL

WolfWalter, 2018-10-17:
Hi @dgarske,
I provisioned the ATECC with the Microchip-aws-iot-zero-touch-secure-provisioning-kit as documented [here](http://microchipdeveloper.com/iot:ztpk) and [here](https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit). I also ported the firmware to the PIC32MZ, which uses the TLS-stack from the WINC1500 wifi-module. That also worked without problems.
As a base I used the now working awsiot example and did the following steps to use certificates the provisioned ATECC:
* Loaded and used the mqtt parameters from the ATECC (URL, client id)
* Loaded the AWS root-CA with `wolfSSL_CTX_load_verify_buffer` (this is not needed for the WINC1500 stack)
* Loaded the certs from the ATECC (bases on the implementation in the zero provisioning firmware) with `wolfSSL_CTX_use_certificate_chain_buffer`. The Wireshark log of the client TLS-"Certificate" package looks identical to the WINC1500 implementation, so I guess this part works
* Limited the cipher list to "ECDHE-ECDSA-AES128-GCM-SHA256:ECDHE-ECDSA-AES128-SHA256" with wolfSSL_CTX_set_cipher_list

At runtime the `atcatls_verify_signature_cb` fails with ATCA_CHECKMAC_VERIFY_FAILED in `atcab_verify_extern`. I added some logging to the `atcatls_verify_signature_cb` function directly before the `atmel_ecc_verify` call.
[ws.zip](https://github.com/wolfSSL/wolfssl/files/2487523/ws.zip)
[console.txt](https://github.com/wolfSSL/wolfssl/files/2487524/console.txt)

The AWS-server sends two certificates. The content of `sig` is identical to the signature in the TLS-"server key exchange" in wireshark for both verifications. The `peerKey` is the same as in the public key for each certficate. I guess I have to study the TLS-standard and the wolfssl documentation to get a better understanding if this is correct. Maybe you could give some clarification?

When I comment the hardware verification out and use wc_ecc_make_key_ex the atcatls_sign_certificate_cb leads to an TLS "Bad Certificate" alert.

dgarske, 2018-10-19:
Hi @WolfWalter,

You might have success with some recent fixes in PR #1885 here:
https://github.com/wolfSSL/wolfssl/pull/1885

It resolves a few issues around using PK callbacks, key size selection and hash algorithms. Let me know if that resolves any of the issues you are seeing.

Thanks,
David Garske, wolfSSL

WolfWalter, 2018-10-22:
Hi @dgarske,
with this PR the used signature for the second certificate changes, but it does not resolve the issue:
[console2.txt](https://github.com/wolfSSL/wolfssl/files/2500425/console2.txt)
[ws2.zip](https://github.com/wolfSSL/wolfssl/files/2500426/ws2.zip)

WolfWalter, 2018-10-22:
Hi @dgarske,
the PR did not resolve my issues. The used signature for the second certificate changed. Here are the log files:
[console2.txt](https://github.com/wolfSSL/wolfssl/files/2500614/console2.txt)
[ws2.zip](https://github.com/wolfSSL/wolfssl/files/2500615/ws2.zip)

WolfWalter, 2018-10-26:
Hi @dgarske,
I think I found the issue! `sigRs` is excpected to be 64 Byte by `atmel_ecc_sign` and `atmel_ecc_verify` but was 128 Byte. My use case now works 100%. I submitted a  [PR](https://github.com/wolfSSL/wolfssl/pull/1897).

Thanks again for the support.

closed 2018-10-26 by WolfWalter