wolfSSL/wolfssl #1807: How to make .a file ?

Vaibhav1587, 2018-09-04:
I wanted to know, how to make .a lib for wolfssl only. I tried as per my knowledge but, getting .la file.

Please guide me how do i generate .a file use it in my own project (linux based).
So that I can use it with windows machine for wolfssh implementation for STM32.

ejohnstown, 2018-09-04:
If you are using "configure" to set up your build, you need to add "--disable-shared". This option will force the build of the static library and disable the dynamic library. It will produce the library archive file.

ejohnstown, 2018-09-13:
Which toolchain/IDE are you using?

Vaibhav1587, 2018-09-14:
I'm using Atollic TrueSTUDIO IDE.

ejohnstown, 2018-09-14:
It sounds like you got it building over in that wolfSSH Issue. I'm going to close this one.

closed 2018-09-14 by ejohnstown