wolfSSL/wolfssl #1812: EVP_BytesToKey error

zero-rp, 2018-09-06:
EVP_BytesToKey(EVP_rc4(), EVP_md5()) No return

ejohnstown, 2018-09-06:
What do you mean by "no return"? Is EVP_BytesToKey() returning a 0 for you?

zero-rp, 2018-09-07:
key and iv No return

ejohnstown, 2018-09-07:
For that function the return code 0 is an error condition. Have you enabled RC4 and MD5? We disable those algorithms by default as they are not considered secure and should not be used. You have to explicitly enable them.

ejohnstown, 2019-02-22:
I apologize for the delay in a response. You are not getting a response from the function because we have not implemented RC4 support for the function `EVP_BytesToKey()`. RC4 is not secure and we don't recommend ever using it.

closed 2019-02-22 by ejohnstown

ejohnstown, 2019-02-22:
As a follow up, when I run the test program, which uses AES128-CBC and SHA-256:

    #include <stdio.h>
    #include <wolfssl/options.h>
    #include <wolfssl/ssl.h>

    static void DumpOctetString(byte* input, int sz);

    int main(void)
    {
        const unsigned char salt[8] = { 1,2,3,4,5,6,7,8 };
        const char data[] = "sphygmomanometer";
        unsigned char key[AES_128_KEY_SIZE];
        unsigned char iv[AES_128_KEY_SIZE];
        int size;

        wolfSSL_Init();

        size = EVP_BytesToKey(EVP_aes_128_cbc(), EVP_sha256(),
                salt, NULL, 0, 10000, key, iv);
        printf("size = %d\n", size);
        size = EVP_BytesToKey(EVP_aes_128_cbc(), EVP_sha256(),
                salt, (const unsigned char*)data, strlen(data),
                10000, key, iv);
        printf("key:\n");
        DumpOctetString(key, AES_128_KEY_SIZE);
        printf("iv:\n");
        DumpOctetString(iv, AES_128_KEY_SIZE);

        wolfSSL_Cleanup();

        return 0;
    }

I get the output:

    size = 16
    key:
    0000: E0 34 7A C8 98 1A 0C 37 E2 75 77 B3 11 1D 32 AA
    iv:
    0000: B5 12 AB EC 72 FE C7 AE 73 5D 35 F1 EC 7E 4D 7E