wolfSSL/wolfssl #1814: GCM output does not match normal AES operation?

mateoconlechuga, 2018-09-06:
Is there a reason why the following code listing does not output the same encryption?

```c
void print_array(byte *b, unsigned int sz)
{
    unsigned int n;
    for( n = 0; n < sz; n++ )
    {
        fprintf(stdout, "%02x ", b[n]);
    }
    fprintf(stdout, "\r\n");
}

int main(void)
{
    Aes enc1;
    Aes enc2;

    byte aes_key[16];
    byte aes_cipher[16];
    byte aes_gcm_cipher[16];
    byte aes_tag[16];
    byte zeros[16];

    memset(zeros, 0, 16);
    memset(aes_key, 0, 16);

    wc_AesSetKeyDirect(&enc1, (const byte*)aes_key, 16, NULL, AES_ENCRYPTION);
    wc_AesGcmSetKey(&enc2, (const byte*)aes_key, 16);

    wc_AesEncryptDirect(&enc1, aes_cipher, zeros);
    wc_AesGcmEncrypt(&enc2, aes_gcm_cipher, zeros, 16, zeros, 16, aes_tag, 16, NULL, 0);

    fprintf(stdout, "original:   ");
    print_array(zeros, 16);
    fprintf(stdout, "cipher:     ");
    print_array(aes_cipher, 16);
    fprintf(stdout, "gcm cipher: ");
    print_array(aes_gcm_cipher, 16);
    
    return 0;
}
```

Even though GCM has an extra XOR, the XOR is against zeros, and should still result in the same encrypted data. I'm a little confused on why the outputs are not identical? Thanks!

Here's the output from the program:

```
original:   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
cipher:     66 e9 4b d4 ef 8a 2c 3b 88 4c fa 59 ca 34 2b 2e 
gcm cipher: a3 b2 2b 84 49 af af bc d6 c0 9f 2c fa 9d e2 be 
```

ejohnstown, 2018-09-06:
You should get different cipher texts. AES-CBC and AES-GCM are completely different.

GCM is a counter mode. A counter value block is encrypted and then XORed into the plaintext to make the cipher text.

CBC takes an encrypted block, XORs it into the next plain block, then encrypts that block, and repeats. The IV is used as the first encrypted block.

mateoconlechuga, 2018-09-06:
Yes, and what I am saying is that with a single block like shown above, they should be identical. 

In GCM, The counter value block is all zeros, and then XOR'd against zeros which should result in the encrypted counter value block.

In Direct Mode, there is exactly a single block of zeros encrypted. 

So I am wondering even though they are different modes, they should have the same resultant data?

mateoconlechuga, 2018-09-06:
Replacing the core with this may make it easier to see:

```c
    wc_AesSetKey(&enc1, (const byte*)aes_key, 16, zeros, AES_ENCRYPTION);
    wc_AesGcmSetKey(&enc2, (const byte*)aes_key, 16);

    wc_AesCtrEncrypt(&enc1, aes_cipher, zeros, 16);
    wc_AesGcmEncrypt(&enc2, aes_gcm_cipher, zeros, 16, zeros, 16, aes_tag, 16, NULL, 0);
```

AES-CTR and AES-GCM are very closely related; GCM just uses an extra GHASH.

ejohnstown, 2018-09-06:
Per NIST SP 800-38D, Algorithm 4, step 2: If len(IV) ≠ 96, then let s = 128 ⎡len(IV)/128⎤- len(IV), and let J0=GHASH(H, IV||0||[len(IV)]). (The text didn't copy here very well.)

The length of your IV is 128, so it gets GHASHed before getting encrypted with AES and then XORed.

mateoconlechuga, 2018-09-06:
Ah that makes a lot of sense. Changing it to this now results in identical output:

```c
    byte iv[16];
    memset(iv, 0, 16);
    iv[15] = 2;

    wc_AesSetKey(&enc1, (const byte*)aes_key, 16, iv, AES_ENCRYPTION);
    wc_AesGcmSetKey(&enc2, (const byte*)aes_key, 16);

    wc_AesCtrEncrypt(&enc1, aes_cipher, zeros, 16);
    wc_AesGcmEncrypt(&enc2, aes_gcm_cipher, zeros, 16, zeros, 12, aes_tag, 16, NULL, 0);
```

Thanks for the help!

closed 2018-09-06 by mateoconlechuga