wolfSSL/wolfssl #1837: Atmel CryptoAuthLib driver for ATECC508A no more compatible

Georges760, 2018-09-17:
Please note you are using the atcatls API of CryptoAuthLib in your wolfcrypt/src/port/atmel/atmel.c, and it's support was discontinued on March 30th : https://github.com/MicrochipTech/cryptoauthlib/commit/fdf5ba686c7432c03b1fd85271a9fadb209dd39e

dgarske, 2018-09-17:
Hi Georges760, that is correct and is something we are working on fixing right now. The results will make it into PR #1815 shortly. Thanks for the report! 
David Garske, wolfSSL

dgarske, 2018-10-16:
Hi @Georges760 : The atcatls dependencies have been eliminated and we support using the latest CryptoAuth lib with changes in PR #1815 . Looking forward to your feedback!

Thanks,
David Garske, wolfSSL

dgarske, 2018-10-29:
Hi @Georges760 : A fix has been merged for ATECC508A/608A that resolves an issue with the reference PK callbacks signature size. Marking this ticket as solved.

closed 2018-10-29 by dgarske