wolfSSL/wolfssl #1858: wolfssl example client and server cannot establish TLS 1.3 connection with early data

muelli, 2018-10-01:
I've tried to use the example client and server for establishing a TLS1.3 connection.

```
git clone https://github.com/wolfSSL/wolfssl.git wolfssl
cd wolfssl
./autogen.sh
./configure --enable-debug --enable-tls13 --enable-session-ticket --enable-sniffer --enable-earlydata
make -j4
./examples/server/server  -v4 &
./examples/client/client -v4
```

Here is my log:

```
➜ tmp>git clone https://github.com/wolfSSL/wolfssl.git wolfssl
Cloning into 'wolfssl'...
remote: Enumerating objects: 61841, done.
remote: Total 61841 (delta 0), reused 0 (delta 0), pack-reused 61841
Receiving objects: 100% (61841/61841), 58.56 MiB | 1.37 MiB/s, done.
Resolving deltas: 100% (45567/45567), done.
➜ tmp>cd wolfssl/

➜ wolfssl git:(master)>./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: creating directory build-aux
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:21: installing 'build-aux/compile'
configure.ac:23: installing 'build-aux/config.guess'
configure.ac:23: installing 'build-aux/config.sub'
configure.ac:26: installing 'build-aux/install-sh'
configure.ac:26: installing 'build-aux/missing'
Makefile.am: installing 'build-aux/depcomp'
parallel-tests: installing 'build-aux/test-driver'
autoreconf: Leaving directory `.'

➜ wolfssl git:(master)>./configure --enable-debug --enable-tls13 --enable-session-ticket --enable-sniffer --enable-earlydata
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for style of include used by make... GNU
checking whether make supports nested variables... yes
[... 579 lines cut ...]

With a bit of debugging, it turns out to error out in src/internal.c with BAD_KEY_SHARE_DATA:
```
#ifdef WOLFSSL_TLS13
        if (IsAtLeastTLSv1_3(ssl->version) &&
            ssl->options.side == WOLFSSL_SERVER_END) {
            /* Try to establish a key share. */
            int ret = TLSX_KeyShare_Establish(ssl);
            if (ret == KEY_SHARE_ERROR)
                ssl->options.serverState = SERVER_HELLO_RETRY_REQUEST_COMPLETE;
            else if (ret != 0) {
                fprintf (stderr, "TLSX_KeyShare_Establish: %d\n", ret);
                return 0;
            }
        }
        else if (first == TLS13_BYTE) {
            /* Can't negotiate TLS 1.3 ciphersuites with lower protocol
             * version. */
            WOLFSSL_MSG("TLS13_BYTE");
            return 0;
        }
#endif
```

I expected to be able to establish a TLS 1.3 with the provided client and server.

dgarske, 2018-10-01:
Hi muelli,

I believe Early data requires Pre-Shared-Keys (PSK) or Session Tickets. Our engineer Sean (@SparkiDev) will followup with further details.

`AC_MSG_ERROR([cannot enable earlydata without enabling session tickets and/or PSK.])`

Did you try using the `-0` option to enable early data with the example client?

An interesting blog post we did regarding TLS v1.3 early data:
https://www.wolfssl.com/use-0-rtt-rope-climb-without-hanging/

Thanks,
David Garske, wolfSSL

SparkiDev, 2018-10-01:
Hi muelli,

I believe that the 'sniffer' configuration item is causing issues.
This option is not supported with TLS 1.3 at this time.

Sean Parkinson, wolfSSL

muelli, 2018-10-02:
Hi Sean,

configuring without the sniffer doesn't help. Configuring without the early data helps, though.
But I don't think it should fail when the server and the client are run with `-r` for `resume` and `-0`.

In any case, some form of message about incompatibility would be nice rather than just failing.

SparkiDev, 2018-10-07:
Hi muelli,

I'm really confused with this. I have tried your exact configuration line without --enable-sniffer and the server and client work.
Can you try again with the latest master?

Sean Parkinson, wolfSSL

SparkiDev, 2018-10-08:
Thanks muelli for taking the time to report your issue.
For me, the sniffer configuration option caused the problem that you reported.

Please don't hesitate to report any issues you find in the future.

Sean Parkinson, wolfSSL

closed 2018-10-08 by SparkiDev

muelli, 2018-10-09:
so is it not a bug that you compile wolfssl s.t. the examples don't work?

SparkiDev, 2018-10-09:
It is not a bug that the --enable-sniffer doesn't work.
This is a feature not implemented.

The sniffer code processes handshake messages. The TLS 1.3 messages are not supported at this time. This is a feature that can be requested but not a bug.

My apologies for not seeing the title change of the issue.

Sean Parkinson, wolfSSL