wolfSSL/wolfssl #1864: WolfSSL fails to connect to ddocdp.do.motive.com with error -328 malformed buffer input error

pelya, 2018-10-03:
WolfSSL 3.15.3, built on Debian 9 x64 with default options.

    examples/client/client -h ddocdp.do.motive.com -p 443 -d
    wolfSSL_connect error -328, malformed buffer input error
    wolfSSL error: wolfSSL_connect failed

OpenSSL connects to this host with no issues (please ignore HTTP error, TLS phase works fine):

    curl https://ddocdp.do.motive.com:443/ -k
    <html><head><title>JBWEB000065: HTTP Status 405 - HTTP method GET is not supported by this URL</title> ...

Wireshark captures of error and success scenarios:
[wolfssl-3.15.3-malformed-buffer-input-error.pcapng.gz](https://github.com/wolfSSL/wolfssl/files/2443289/wolfssl-3.15.3-malformed-buffer-input-error.pcapng.gz)
[openssl-success.pcapng.gz](https://github.com/wolfSSL/wolfssl/files/2443290/openssl-success.pcapng.gz)

kaleb-himes, 2018-10-04:
@pelya,

Thank you so much for this report! I can confirm we can reproduce this. We are looking into it and will get you an update as soon as possible!

- K

kaleb-himes, 2018-10-08:
@pelya ,

Can you tell us a little bit about this domain? A modern browser even behaves curiously when connecting to it IE:

browsing to http://ddocdp.do.motive.com:80 just hangs and continuously sends retransmits
browsing to https://ddocdp.do.motive.com:443 results in:

```
JBWEB000065: HTTP Status 405 - HTTP method GET is not supported by this URL
JBWEB000309: type JBWEB000067: Status report
JBWEB000068: message HTTP method GET is not supported by this URL
JBWEB000069: description JBWEB000125: The specified HTTP method is not allowed for the requested resource.
```

What is this domain and what is it used for?

Warm Regards,

K

pelya, 2018-10-09:
Here is a small test file on this server, it can be downloaded by Firefox
with no issues:

https://ddocdp.do.motive.com/firmware/Sequans_456_789.bin

This domain is used as a file server for a test framework we are using, it
is operated by another company, so I can upload files there but cannot tell
anything else about it. It seems to be Java-based and run some version of
JBoss software.

kaleb-himes, 2018-10-11:
Thank you @pelya for those details and for the target to download. We'll do some more testing and let you know what we find.

- Kaleb

ejohnstown, 2018-10-12:
In pull request #1872 I added an option to allow the client to accept, but ignore, the Support Groups extension from the server hello.

closed 2018-10-12 by ejohnstown