wolfSSL/wolfssl #1875: tls 1.3: the leading zero is lost in the DH(FFDHE) KE Secret

Neilpang, 2018-10-15:
As in tls 1.3 spec:
https://tools.ietf.org/html/rfc8446#section-7.4.1

```
For finite field groups, a conventional Diffie-Hellman [DH76]
   computation is performed.  The negotiated key (Z) is converted to a
   byte string by encoding in big-endian form and left-padded with zeros
   up to the size of the prime. 
```

For FFDHE, the Key exchange Secret length should be same as the  prime size.
wolfssl doesn't  left pad zero to the desired size.

For example, if FFDHE 2048 is used,  the KE secret should always be 256 bytes, but wolfssl just use 255 bytes if it has a leading zero.
example output log:
```
...
        ee f0 83 7d e6 1f 39 6f 13 42 14 fa 40 47 3d 7e | ...}..9o.B..@G=~
KE Secret
        ba 0e 75 c0 7e 84 be 08 f7 03 9c 13 ea d4 c0 19 | ..u.~...........
        bb 3e 71 66 62 f0 34 2c 20 c1 2d d6 21 2d e4 df | .>qfb.4, .-.!-..
        96 78 a7 a7 0b dd a4 7a 7b c2 31 43 8e 0d 3f 02 | .x.....z{.1C..?.
        66 fe 09 4c 9a ac 58 85 5a 32 31 0f 90 6d b3 cd | f..L..X.Z21..m..
        aa b0 75 1a 28 5c c8 be 02 f2 61 0f 62 7e 25 3e | ..u.(\....a.b~%>
        5f 67 1c 79 a7 17 42 67 0b 11 24 4f af d6 01 8f | _g.y..Bg..$O....
        c5 6a fa 8b 80 d5 69 79 f6 fc 5c cb cd 2d 15 e0 | .j....iy..\..-..
        70 fa 03 a3 2a 77 0a eb e4 7c ae e9 15 2d c4 41 | p...*w...|...-.A
        b9 a1 11 d9 d0 5f 9c 9a 84 6b 20 76 c8 ab 9d e4 | ....._...k v....
        2b 1e 31 0b 3d d5 48 9b 12 d9 a1 ca 84 b6 81 e3 | +.1.=.H.........
        a4 41 56 04 37 27 9b ec ec 5b 57 46 71 e4 c5 ee | .AV.7'...[WFq...
        90 82 e3 bd 06 a2 8b 16 ae b2 c8 3d 3c bb 5b 9a | ...........=<.[.
        38 d9 61 5d c5 24 7e 48 7c 4b a0 9d a3 90 41 7c | 8.a].$~H|K....A|
        c2 3a b8 fa 9f 8f d5 5e 6c 2f 07 66 66 c7 d9 88 | .:.....^l/.ff...
        27 f0 27 84 3b d2 d5 7e e8 3c 12 9e ea 07 2b 58 | '.'.;..~.<....+X
        91 f1 88 a0 55 21 1f 3f 60 5a 20 60 e2 93 e0    | ....U!.?`Z `...
accept state SERVER_HELLO_SENT
wolfSSL Entering SendTls13EncryptedExtensions
Derive Handshake Secret
```

The "KE Secret" is only 255 bytes.

Steps to reproduce, just for simplicity, I use WolfSSL example server and GnuTLS-cli to reproduce.

1.  Run wolfSSL example server, listening at 8799 port:

```sh
server.exe  -b -i -d -g  -v 4 -p 8799
```

2. Compile and install current GnuTls  3.6.4, then run `gnutls-cli`:

```sh

while true; do  
  timeout 1  gnutls-cli    -V  --debug=9999 --insecure  --priority "NORMAL:-CURVE-ALL:+VERS-TLS1.3:+GROUP-FFDHE2048:+AES-128-GCM:+SIGN-RSA-SHA256"   192.168.8.1:8799;  
done;

```

3. Just wait from a few seconds or minutes,  the wolfssl example server will crash.

Neilpang, 2018-10-15:
BTW, it''s not possible to reproduce with `openssl-cli`, since openssl doens't support FFDH in tls 1.3 at all.

Neilpang, 2018-10-16:
Hi  @kaleb-himes 
Thanks for  your reply.

It seems this bug was fixed in master branch: https://github.com/wolfSSL/wolfssl/blob/master/src/tls.c#L6403

I tried master code, it's working now.

I was using the current v3.15.3 source code downloaded from the official website: 
https://www.wolfssl.com/download/

The issue was there.

Thanks.

closed 2018-10-16 by Neilpang

kaleb-himes, 2018-10-16:
@Neilpang,

Thank you for confirming the issue is resolved in master! That fix will go out with our next release.

Warmest Regards,

K