wolfSSL/wolfssl #1900: Do not open this url: https://3w.huanqiu.com/a/34f321/7HmWU1ZpF60?agt=8

weolar, 2018-10-28:
I use curl and wolfssl to open https://3w.huanqiu.com/a/34f321/7HmWU1ZpF60?agt=8
show error: Curl_failf:SSL_connect failed with error -313: revcd alert fatal error.
but I change to openssl, it is no problem.

dgarske, 2018-10-29:
Hi weolar,

This server uses Server Name Indicator (SNI) to determine the server to route to. https://en.wikipedia.org/wiki/Server_Name_Indication

This feature is not enabled by default and must be enabled by using `--enable-sni` or `HAVE_SNI`.

Then you must call the API `wolfSSL_CTX_UseSNI` or `wolfSSL_UseSNI` to enable it.

API Example:
```
wolfSSL_CTX_UseSNI(ctx, 0, sniHostName, (word16) XSTRLEN(sniHostName));
```

Client Example:
```
./configure --enable-sni
make
./examples/client/client -p 443 -h 23.236.125.8 -d -x -g -S 3w.huanqiu.com
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
SSL curve name is SECP256R1
SSL connect ok, sending GET...
HTTP/1.1 404 Not Found
Server: openresty+
Date: Mon, 29 Oct 2018 16:55:33 GMT

Content-Type: text/html
Content-Length: 166
Connection: close

<html>
<h
```

I believe if you build wolfSSL with `./configure --enable-sni && make && sudo make install` and rebuild Curl it will work correctly.

Thanks,
David Garske, wolfSSL

weolar, 2018-10-29:
thank you, exactly

closed 2018-10-29 by dgarske

weolar, 2018-10-30:
https://rexl.tk/
hi, I build wolfSSL with HAVE_SNI, but this url show "Curl_failf:SSL_connect failed with error -313: revcd alert fatal error"

dgarske, 2018-10-30:
Hi weolar,

Curl does support SNI with wolfSSL:
https://github.com/curl/curl/blob/master/lib/vtls/cyassl.c#L359

Here is an end-to-end build/test that works for me:

```
git clone https://github.com/wolfSSL/wolfssl.git
cd wolfssl
./autogen.sh
./configure --enable-sni
make
sudo make install

cd ..
git clone https://github.com/curl/curl.git
cd curl
./buildconf 
./configure --with-wolfssl --without-ssl
make
sudo make install

curl https://3w.huanqiu.com/a/34f321/7HmWU1ZpF60?agt=8

<!DOCTYPE html>
<html lang="zh">
<head>
    <meta charset="UTF-8">
...
```

Thanks,
David Garske, wolfSSL