wolfSSL/wolfssl #1919: stack out of bounds read / underflow in MaskMac

hannob, 2018-11-09:
With Address Sanitizer I see an invalid memory access in MaskMac(). I haven't analyzed this in detail.

This happens if I run the example server code on a public host compiled with ASAN and run the SSLLabs test against it.

ASAN error / stack trace:
```
==8929==ERROR: AddressSanitizer: stack-buffer-underflow on address 0x7ffe8285263e at pc 0x000000595405 bp 0x7ffe82852610 sp 0x7ffe82852608
READ of size 1 at 0x7ffe8285263e thread T0
    #0 0x595404 in MaskMac /f/wolf/wolfssl/src/internal.c:12452:33
    #1 0x595404 in TimingPadVerify /f/wolf/wolfssl/src/internal.c:12479
    #2 0x599215 in VerifyMac /f/wolf/wolfssl/src/internal.c:12732:19
    #3 0x599215 in ProcessReply /f/wolf/wolfssl/src/internal.c:13097
    #4 0x5a609b in ReceiveData /f/wolf/wolfssl/src/internal.c:15196:28
    #5 0x4fcc93 in wolfSSL_read_internal /f/wolf/wolfssl/src/ssl.c:1769:11
    #6 0x4fcc93 in wolfSSL_read /f/wolf/wolfssl/src/ssl.c:1816
    #7 0x4f7f64 in ServerRead /f/wolf/wolfssl/examples/server/server.c:292:15
    #8 0x4f7f64 in server_test /f/wolf/wolfssl/examples/server/server.c:1932
    #9 0x4f977e in main /f/wolf/wolfssl/examples/server/server.c:2090:9
    #10 0x7f46793c3ae6 in __libc_start_main (/lib64/libc.so.6+0x21ae6)
    #11 0x41b489 in _start (/f/wolf/wolfssl/examples/server/server+0x41b489)

Address 0x7ffe8285263e is located in stack of thread T0 at offset 30 in frame
    #0 0x5945ff in TimingPadVerify /f/wolf/wolfssl/src/internal.c:12468

  This frame has 2 object(s):
    [32, 96) 'mac.i' (line 12412) <== Memory access at offset 30 underflows this variable
    [128, 192) 'verify' (line 12469)
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-underflow /f/wolf/wolfssl/src/internal.c:12452:33 in MaskMac

dgarske, 2018-11-09:
Hi Hanno,

Thanks for this report! It looks like this change was made commit e684156a1ecff2ef3bd9a57b30f02ab8bca2a4eb on 5/29/18. 

We will take a look at this quickly.

Thanks,
David Garske, wolfSSL

SparkiDev, 2018-11-12:
Thanks Hanno!

I've put up a pull request, #1922, that has this change but in a constant time way.

Sean

--
Sean Parkinson, wolfSSL

SparkiDev, 2018-11-12:
Hi Hanno,

The pull request has been merged into the master.
Once again thank you for reporting this issue!

Sean

--
Sean Parkinson, wolfSSL

closed 2018-11-12 by SparkiDev