wolfSSL/wolfssl #1935: Add several missing OpenSSL APIs

xenuvix, 2018-11-22:
When trying to port my OpenSSL based project to wolfSSL, I found out that the next APIs and symbols are missing.

EVP_aes_256_xts
EVP_CIPHER_name
EVP_MAX_IV_LENGTH

grooverdan, 2020-05-02:
Some other functions https://github.com/openssl/openssl/commit/83b06347023a573433b6aa23c8042f89df869f9e

- EVP_CIPHER_CTX_encrypting()
- EVP_CIPHER_CTX_iv()
- EVP_CIPHER_CTX_iv_noconst()
- EVP_CIPHER_CTX_original_iv()
- EVP_CIPHER_CTX_buf_noconst()
- EVP_CIPHER_CTX_num()
- EVP_CIPHER_CTX_set_num()
- EVP_CIPHER_CTX_cipher_data()
- EVP_CIPHER_CTX_new_cipher_data()

I hit  EVP_CIPHER_CTX_encrypting() which seems to map to `{ return ctx->enc; }`

bgermann, 2020-09-19:
The original request's symbols are available now.

haydenroche5, 2021-01-13:
Our OpenSSL compatibility layer is always growing. If you have additional functions that you'd like us to add, please feel free to reach out. Adding new functions for the sake of completeness is a low priority for us, but we're of course happy to implement new ones if there's a customer need.

closed 2021-01-13 by haydenroche5