wolfSSL/wolfssl #1980: Undefined OpenSSL functions with compatibility layer

DenisLug, 2018-12-13:
I'm getting the error message in my program that all OpenSSL functions are not defined in the scope. The file wolfssl/openssl/ssl.h is included in my program and I've configured and built wolfSSL with the options `--enable-opensslextra` and even `--enable-all`. When I look for the defined symbols in the  libwolfssl.so with `nm src/.libs/libwolfssl.so`, I can't find any OpenSSL function.

Did I miss anything when working with the OpenSSL compatibility layer?

dgarske, 2018-12-13:
Hi DenisLug,

The most likely reason is due to your build options not being included in your application. The wolfSSL `./configure --enable-opensslextra` generates a `./wolfssl/options.h` file that contains your build options including `OPENSSL_EXTRA`. Try adding `#include <wolfssl/options.h>` in your application before including any openssl or wolfssl headers. There is also a possible solution by adding this include in our settings.h like is done in PR #1899. You might also try adding the `--enable-opensslall` option, which exposes all openssl compatibility layer features.

Let me know if that helps. If not please let us know which functions are reporting undefined.

Thanks,
David Garske, wolfSSL

DenisLug, 2018-12-14:
Hi David,

thank you for your help. Everything is working now. It would be nice to add a note to the documentation to add the `#include <wolfssl/options.h>` into the code.

Regards
Denis

dgarske, 2018-12-19:
Hi @DenisLug : Happy to hear things are working. I've updated our user manual, which gets published with each official release.

This issue of not including options.h is also in our Frequently Asked Questions here.
https://www.wolfssl.com/docs/frequently-asked-questions-faq/#How_do_I_manage_the_build_configuration_for_wolfSSL?

Attached is a great presentation we did for a Webinar in Aug 2018 that also covers this on page 10. 
[wolfSSL_Compatibility_Layer.pdf](https://github.com/wolfSSL/wolfssl/files/2695781/wolfSSL_Compatibility_Layer.pdf)

This presentation is also on YouTube here: https://www.youtube.com/watch?v=9IzHPWMnDHo

Thanks,
David Garske, wolfSSL

closed 2018-12-19 by dgarske