wolfSSL/wolfssl #1995: There is no  `wolfSSL_EC_POINT_set_affine_coordinates_GFp` function

miluoalbert, 2018-12-21:
There is a `EC_POINT_set_affine_coordinates_GFp` API function in OPENSSL, but I can't find a `wolfSSL_EC_POINT_set_affine_coordinates_GFp` API in wolfssl. 
Is there any way for me to establish a `EC_POINT` from X,Y coordinates(`WOLFSSL_BIGNUM`) of public point send from my server?

roytam1, 2018-12-26:
just did a `grep` in `wolfssl-master` and found it:
https://github.com/wolfSSL/wolfssl/blob/8c0a55d43bb5a2f8a72ec106bbaa42876e0fd282/src/ssl.c#L27347

dgarske, 2018-12-27:
Hi Miluoalbert,

You are correct the `EC_POINT_set_affine_coordinates` openssl compatibility function is missing. We are constantly expanding and adding new compatibility functions. I've made a note to look into this one. Its always helpful to understand the use case and project. Can you tell us more about your project and timeline? If you would like to email us directly about this please email support@wolfssl.com.

Thanks,
David Garske, wolfSSL

haydenroche5, 2021-01-13:
Our OpenSSL compatibility layer now supports EC_POINT_set_affine_coordinates_GFp. It maps to wolfSSL_EC_POINT_set_affine_coordinates_GFp. Closing this out.

closed 2021-01-13 by haydenroche5