wolfSSL/wolfssl #1999: problem with wolfSSL TLS 1.0 and old clients

roytam1, 2018-12-23:
I tried compiling https://github.com/kaleb-himes/stunnel with wolfSSL-master with TLS 1.0 and RC4 enabled for old clients such as IE6.
and some of XP-based downloader that is using system SChannel library can't connect to my stunnel-wolfssl server and logged `SSL_accept: FFFFFEC9: unknown type in record hdr`. Is there a way to solve/workaround it?

roytam1, 2018-12-24:
OK found one reason (of many) that breaks old client: they send SSLv2 ClientHello packets. So I need to add `-DOLD_HELLO_ALLOWED` into CFLAGS.

Another connection problem happens on POSReady2009 machine which is Windows XP based system with its schannel library is TLS 1.2 and TLS_RSA_WITH_AES_256_CBC_SHA256 capable, for unknown reason, schannel client doesn't accept data from wolfssl after Hello done.

zipped pcap file attached (I have to rename as docx in order to able to upload)
[posready2009-ie8-tls12.pcap.zip](https://github.com/wolfSSL/wolfssl/files/2707605/posready2009-ie8-tls12.docx)

roytam1, 2018-12-25:
So it seems to be cert problem, signing new cert with SHA1 make progress go a bit further, but now stunnel throws another problem:

````
2018.12.25 20:42:49 LOG3[2]: error queue: FFFFFECF: verify mac problem
2018.12.25 20:42:49 LOG3[2]: SSL_accept: FFFFFEC8: error during decryption
````

zipped pcap file attached (I have to rename as docx in order to able to upload)
[tls12-schannel-sha11cert.zip](https://github.com/wolfSSL/wolfssl/files/2708759/tls12-schannel-sha11cert.docx)

roytam1, 2018-12-25:
debug log: (from WinXP IE8 TLS 1.2):
````
2018.12.25 21:00:36 LOG5[0]: Service [stunnel3] accepted connection from 192.168.168.203:48050
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SSL_new
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_set_ex_data
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SSL_set_fd
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SSL_set_read_fd
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SSL_set_write_fd
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_set_accept_state
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SSL_accept()
2018.12.25 21:00:36 LOG4[0]: growing input buffer.
2018.12.25 21:00:36 LOG4[0]: received record layer msg
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering DoHandShakeMsg()
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering DoHandShakeMsgType
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_get_ex_data
2018.12.25 21:00:36 LOG4[0]: processing client hello
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering DoClientHello
2018.12.25 21:00:36 LOG4[0]: Matched No Compression
2018.12.25 21:00:36 LOG4[0]: Adding signature algorithms extension
2018.12.25 21:00:36 LOG4[0]: Signature Algorithms extension received
2018.12.25 21:00:36 LOG4[0]: Extended Master Secret extension received
2018.12.25 21:00:36 LOG4[0]: Secure Renegotiation extension received
2018.12.25 21:00:36 LOG4[0]: Calling custom sni callback
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering MatchSuite
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering VerifyServerSuite
2018.12.25 21:00:36 LOG4[0]: Requires RSA
2018.12.25 21:00:36 LOG4[0]: Verified suite validity
2018.12.25 21:00:36 LOG4[0]: Shrinking input buffer.
2018.12.25 21:00:36 LOG4[0]: accept state ACCEPT_FIRST_REPLY_DONE
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SendServerHello
2018.12.25 21:00:36 LOG4[0]: growing output buffer.
2018.12.25 21:00:36 LOG4[0]: EMS extension to write
2018.12.25 21:00:36 LOG4[0]: Shrinking output buffer.
2018.12.25 21:00:36 LOG4[0]: accept state SERVER_HELLO_SENT
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SendCertificate
2018.12.25 21:00:36 LOG4[0]: growing output buffer.
2018.12.25 21:00:36 LOG4[0]: Shrinking output buffer.
2018.12.25 21:00:36 LOG4[0]: accept state CERT_SENT
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SendCertificateStatus
2018.12.25 21:00:36 LOG4[0]: accept state CERT_STATUS_SENT
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SendServerKeyExchange
2018.12.25 21:00:36 LOG4[0]: accept state KEY_EXCHANGE_SENT
2018.12.25 21:00:36 LOG4[0]: accept state CERT_REQ_SENT
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SendServerHelloDone
2018.12.25 21:00:36 LOG4[0]: growing output buffer.
2018.12.25 21:00:36 LOG4[0]: Shrinking output buffer.
2018.12.25 21:00:36 LOG4[0]: accept state SERVER_HELLO_DONE
2018.12.25 21:00:36 LOG4[0]: growing input buffer.
2018.12.25 21:00:36 LOG4[0]: received record layer msg
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering DoHandShakeMsg()
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering DoHandShakeMsgType
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_get_ex_data
2018.12.25 21:00:36 LOG4[0]: processing client key exchange
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering DoClientKeyExchange
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering RsaDec
2018.12.25 21:00:36 LOG4[0]: preMasterSecret version mismatch
2018.12.25 21:00:36 LOG4[0]: Shrinking input buffer.
2018.12.25 21:00:36 LOG4[0]: received record layer msg
2018.12.25 21:00:36 LOG4[0]: got CHANGE CIPHER SPEC
2018.12.25 21:00:36 LOG4[0]: growing input buffer.
2018.12.25 21:00:36 LOG4[0]: VerifyMac failed
2018.12.25 21:00:36 LOG3[0]: wolfSSL error occurred, error = 305 line:13165 file:src/internal.c
2018.12.25 21:00:36 LOG3[0]: wolfSSL error occurred, error = 312 line:9570 file:src/ssl.c
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SSL_get_error
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_get_error
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_peek_error_line_data
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_get_error
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_peek_error_line_data
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_get_error
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_peek_error_line_data
2018.12.25 21:00:36 LOG4[0]: No Errors in queue
2018.12.25 21:00:36 LOG4[0]: Issue peeking at error node in queue
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_error_string_n
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering ERR_error_string
2018.12.25 21:00:36 LOG3[0]: error queue: FFFFFECF: verify mac problem
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_ERR_error_string_n
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering ERR_error_string
2018.12.25 21:00:36 LOG3[0]: SSL_accept: FFFFFEC8: error during decryption
2018.12.25 21:00:36 LOG5[0]: Connection reset: 0 byte(s) sent to SSL, 0 byte(s) sent to socket
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_set_shutdown
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering SSL_free
2018.12.25 21:00:36 LOG4[0]: CTX ref count not 0 yet, no free
2018.12.25 21:00:36 LOG4[0]: Shrinking input buffer.
2018.12.25 21:00:36 LOG5[0]: wolfSSL Entering wolfSSL_BIO_free
````

roytam1, 2018-12-26:
> cross-reference:
> https://github.com/openssl/openssl/blob/OpenSSL_1_0_0-stable/ssl/t1_enc.c#L871
> https://github.com/openssl/openssl/blob/OpenSSL_1_0_0-stable/ssl/s3_cbc.c#L108

tried hacking wolfssl so it don't serve AES CBC but same error happens, so it is not the fault of dealing with AES CBC padding?

roytam1, 2018-12-26:
may reference this: https://www.wolfssl.com/forums/topic1014-ie11edge-padcheck-failed.html
just the problem is not in PadCheck.

roytam1, 2018-12-27:
tried to debug this, adding these debug code into internal.c:
````
--- ../wolfssl-git/wolfssl-master/src/internal.c        2018-12-27 07:09:42.000000000 +0800
+++ src/internal.c      2018-12-27 21:44:35.000000000 +0800
@@ -12429,29 +12429,58 @@
 {
     byte verify[WC_MAX_DIGEST_SIZE];
     byte good;
     int  ret = 0;

+    {
+        char debug[256];
+        XSNPRINTF(debug, sizeof(debug), "input data, pLen = %d, macSz = %d:", pLen, macSz);
+        WOLFSSL_MSG(debug);
+        WOLFSSL_BUFFER(input,pLen);
+    }
+
+
     good = MaskPadding(input, pLen, macSz);
+    {
+        char debug[256];
+        XSNPRINTF(debug, sizeof(debug), "MaskPadding(first run) good = %d", good);
+        WOLFSSL_MSG(debug);
+    }
     /* 4th argument has potential to underflow, ssl->hmac function should
      * either increment the size by (macSz + padLen + 1) before use or check on
      * the size to make sure is valid. */
     ret = ssl->hmac(ssl, verify, input, pLen - macSz - padLen - 1, padLen,
                                                                     content, 1);
+    {
+        char debug[256];
+        XSNPRINTF(debug, sizeof(debug), "hmac ret = %d", ret);
+        WOLFSSL_MSG(debug);
+    }
     good |= MaskMac(input, pLen, ssl->specs.hash_size, verify);
+    {
+        char debug[256];
+        XSNPRINTF(debug, sizeof(debug), "MaskMac(after hmac) good = %d", good);
+        WOLFSSL_MSG(debug);
+    }

     /* Non-zero on failure. */
     good = (byte)~(word32)good;
     good &= good >> 4;
     good &= good >> 2;
     good &= good >> 1;
     /* Make ret negative on masking failure. */
     ret -= 1 - good;

     /* Treat any faulure as verify MAC error. */
-    if (ret != 0)
+    if (ret != 0) {
+    {
+        char debug[256];
+        XSNPRINTF(debug, sizeof(debug), "TimingPadVerify ret = %d", ret);
+        WOLFSSL_MSG(debug);
+    }
         ret = VERIFY_MAC_ERROR;
+    }

     return ret;
 }
 #endif

@@ -12690,12 +12719,14 @@
         padByte = 1;

         if (ssl->options.tls) {
             ret = TimingPadVerify(ssl, input, pad, digestSz, msgSz - ivExtra,
                                   content);
[... 114 lines cut ...]
2018.12.27 21:47:34 LOG4[0]: .33 61 e0 8d a1 53 1c 5e d5 b8 12 cf 03 59 22 57 | 3a...S.^.....Y"W
2018.12.27 21:47:34 LOG4[0]: Data received
2018.12.27 21:47:34 LOG4[0]: .16 03 03 00 50                                  | ....P
2018.12.27 21:47:34 LOG4[0]: growing input buffer.
2018.12.27 21:47:34 LOG4[0]: Data received
2018.12.27 21:47:34 LOG4[0]: .3e e3 df bf 9d 1a 75 9e 38 5b 49 ca 74 d7 79 cf | >.....u.8[I.t.y.
2018.12.27 21:47:34 LOG4[0]: .3c 21 6d 9c f6 66 3e e7 1b 87 96 ee 66 28 6e 80 | <!m..f>.....f(n.
2018.12.27 21:47:34 LOG4[0]: .ea ea 9e 6a 61 a1 66 ea 80 a6 bc f6 fb 94 32 9d | ...ja.f.......2.
2018.12.27 21:47:34 LOG4[0]: .28 c9 87 62 ab a5 bd 47 d9 1c fc bd dd 70 23 7b | (..b...G.....p#{
2018.12.27 21:47:34 LOG4[0]: .96 d1 2c 89 44 73 da 4b 5f 88 55 b1 0b 48 b3 26 | ..,.Ds.K_.U..H.&
2018.12.27 21:47:34 LOG4[0]: input data, pLen = 64, macSz = 32:
2018.12.27 21:47:34 LOG4[0]: .8f 30 a2 83 f4 68 49 e2 0a 9a 7e 71 fc 6d b9 3f | .0...hI...~q.m.?
2018.12.27 21:47:34 LOG4[0]: .26 d8 10 03 e8 c3 0b 35 40 e0 41 a2 13 88 41 04 | &......5@.A...A.
2018.12.27 21:47:34 LOG4[0]: .6f 8d 28 33 1e e7 1c ae 3c 4a 37 1b d1 d2 7e 7b | o.(3....<J7...~{
2018.12.27 21:47:34 LOG4[0]: .82 e5 8f 5c 7c bc 2f 06 d3 a2 bb 6e 60 e7 73 e2 | ...\|./....n`.s.
2018.12.27 21:47:34 LOG4[0]: MaskPadding(first run) good = 255
2018.12.27 21:47:34 LOG4[0]: hmac ret = 0
2018.12.27 21:47:34 LOG4[0]: MaskMac(after hmac) good = 255
2018.12.27 21:47:34 LOG4[0]: TimingPadVerify ret = -1
2018.12.27 21:47:34 LOG4[0]: TimingPadVerify failed
2018.12.27 21:47:34 LOG4[0]: VerifyMac failed
2018.12.27 21:47:34 LOG3[0]: wolfSSL error occurred, error = 305 line:13107 file:src/internal.c
2018.12.27 21:47:34 LOG3[0]: wolfSSL error occurred, error = 312 line:9570 file:src/ssl.c
````
of course I have no clue why `MaskPadding` fails (if inline comment is correctly describing what function does)

roytam1, 2018-12-27:
and from IE6 client:
````
2018.12.27 22:00:41 LOG4[27]: Data received
2018.12.27 22:00:41 LOG4[27]: .10 00 02 02 02 00 c3 0a 5d 93 a3 27 65 fc fa c9 | ........]..'e...
2018.12.27 22:00:41 LOG4[27]: .21 e1 e0 d4 8c 6c 8c 01 60 30 3e 4f 96 26 dd eb | !....l..`0>O.&..
2018.12.27 22:00:41 LOG4[27]: .35 e5 41 e3 84 89 8f 58 f6 97 88 41 69 74 d3 5e | 5.A....X...Ait.^
2018.12.27 22:00:41 LOG4[27]: .a5 3a 19 af 5a 02 7b ff a4 c7 a2 73 e3 fd 52 11 | .:..Z.{....s..R.
2018.12.27 22:00:41 LOG4[27]: .b2 65 28 71 8b 5e b9 5f 92 ba 2c 7e 3b e3 e8 0b | .e(q.^._..,~;...
2018.12.27 22:00:41 LOG4[27]: .14 b1 46 5e e2 fe e6 fa 69 15 a4 8d ee ab ec b9 | ..F^....i.......
2018.12.27 22:00:41 LOG4[27]: .46 9e 39 4b 50 ab d7 a8 65 b4 15 c4 1e f7 27 c8 | F.9KP...e.....'.
2018.12.27 22:00:41 LOG4[27]: .fa b5 f1 6b 4e 3e d4 3b fe 79 19 3c e2 48 ff 08 | ...kN>.;.y.<.H..
2018.12.27 22:00:41 LOG4[27]: .2d 2b 7c 5c d0 31 28 c2 20 05 43 0e 05 7e c4 db | -+|\.1(. .C..~..
2018.12.27 22:00:41 LOG4[27]: .39 4e 0c 98 92 54 bb 36 7f e2 46 68 6c 03 40 cc | 9N...T.6..Fhl.@.
2018.12.27 22:00:41 LOG4[27]: .4c a2 1c 0b 74 2c e1 90 0a 7e f1 7e da c2 70 4e | L...t,...~.~..pN
2018.12.27 22:00:41 LOG4[27]: .d5 ec 31 cf c3 60 57 22 69 ee c6 c3 a6 3c b0 62 | ..1..`W"i....<.b
2018.12.27 22:00:41 LOG4[27]: .75 cc 87 5c ae cd 78 a9 8d d0 34 5a 98 2b 31 03 | u..\..x...4Z.+1.
2018.12.27 22:00:41 LOG4[27]: .ca 82 b2 ef 36 94 30 95 2f 26 37 3d e0 bd 50 96 | ....6.0./&7=..P.
2018.12.27 22:00:41 LOG4[27]: .99 63 bb df f7 a7 f3 f7 13 8c ac 22 86 db db 5f | .c........."..._
2018.12.27 22:00:41 LOG4[27]: .9f 81 4f be b2 84 80 87 16 6d e7 02 51 ed 1d 26 | ..O......m..Q..&
2018.12.27 22:00:41 LOG4[27]: .55 bb 98 5a cb f5 b8 d5 b3 e1 af 1d 5a 8e c4 67 | U..Z........Z..g
2018.12.27 22:00:41 LOG4[27]: .41 59 65 0c f1 31 94 52 72 2e 9c 90 46 51 d5 a0 | AYe..1.Rr...FQ..
2018.12.27 22:00:41 LOG4[27]: .22 33 cc cd a0 47 10 85 fc 5a 92 cf 5d 5d d8 27 | "3...G...Z..]].'
2018.12.27 22:00:41 LOG4[27]: .4c d4 14 37 de 29 ea fa f6 b1 94 06 42 14 b5 fe | L..7.)......B...
2018.12.27 22:00:41 LOG4[27]: .aa cb db 20 7b 43 2c 2c 8a 0a cc b9 b2 cb 2a 95 | ... {C,,......*.
2018.12.27 22:00:41 LOG4[27]: .49 1f 22 04 c1 41 6e de a3 38 bc 5a e1 3f 3c c0 | I."..An..8.Z.?<.
2018.12.27 22:00:41 LOG4[27]: .24 00 ac c4 c9 4b 19 ab 48 54 72 5c f1 e5 33 47 | $....K..HTr\..3G
2018.12.27 22:00:41 LOG4[27]: .7d 10 b3 64 ff 33 98 b7 5a ac 60 cf 98 60 4a e0 | }..d.3..Z.`..`J.
2018.12.27 22:00:41 LOG4[27]: .fc bb ec 0a c2 77 78 30 76 dd cb e2 d5 7c 40 cb | .....wx0v....|@.
2018.12.27 22:00:41 LOG4[27]: .54 2a 1c 1a fd a7 e7 21 f2 c4 be 93 e4 fd a9 5e | T*.....!.......^
2018.12.27 22:00:41 LOG4[27]: .fe 7a db ed 1b 06 0c d2 c3 c8 26 e0 57 fc c4 3f | .z........&.W..?
2018.12.27 22:00:41 LOG4[27]: .2e 7b 23 20 28 da 9e d7 48 8e 12 f2 31 48 92 64 | .{# (...H...1H.d
2018.12.27 22:00:41 LOG4[27]: .7b 61 2e 4a 8c 20 6c 6a e5 87 d4 75 0d ab 33 4d | {a.J. lj...u..3M
2018.12.27 22:00:41 LOG4[27]: .7d b3 65 b1 52 a4 6f 93 b9 44 66 65 10 05 35 dd | }.e.R.o..Dfe..5.
2018.12.27 22:00:41 LOG4[27]: .88 d1 c8 3b c2 59 c6 28 62 86 de da f7 2c 56 30 | ...;.Y.(b....,V0
2018.12.27 22:00:41 LOG4[27]: .ac 1c 40 ab db a8 b4 1c d7 3e a6 24 8a 68 72 c7 | ..@......>.$.hr.
2018.12.27 22:00:41 LOG4[27]: .76 f3 ed 35 a6 6f                               | v..5.o
2018.12.27 22:00:41 LOG4[27]: received record layer msg
2018.12.27 22:00:41 LOG5[27]: wolfSSL Entering DoHandShakeMsg()
2018.12.27 22:00:41 LOG5[27]: wolfSSL Entering DoHandShakeMsgType
2018.12.27 22:00:41 LOG5[27]: wolfSSL Entering wolfSSL_get_ex_data
2018.12.27 22:00:41 LOG4[27]: processing client key exchange
2018.12.27 22:00:41 LOG5[27]: wolfSSL Entering DoClientKeyExchange
2018.12.27 22:00:41 LOG5[27]: wolfSSL Entering RsaDec
2018.12.27 22:00:41 LOG4[27]: preMasterSecret version mismatch
2018.12.27 22:00:41 LOG4[27]: Shrinking input buffer.
2018.12.27 22:00:41 LOG4[27]: Data received
2018.12.27 22:00:41 LOG4[27]: .14 03 01 00 01                                  | .....
2018.12.27 22:00:41 LOG4[27]: Data received
2018.12.27 22:00:41 LOG4[27]: .01                                              | .
2018.12.27 22:00:41 LOG4[27]: received record layer msg
2018.12.27 22:00:41 LOG4[27]: got CHANGE CIPHER SPEC
2018.12.27 22:00:41 LOG4[27]: Provisioning DECRYPT key
2018.12.27 22:00:41 LOG4[27]: .08 1c b8 ec a5 3e 42 ef c3 41 fe 82 a4 97 94 6f | .....>B..A.....o
2018.12.27 22:00:41 LOG4[27]: .00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
2018.12.27 22:00:41 LOG4[27]: Data received
2018.12.27 22:00:41 LOG4[27]: .16 03 01 00 24                                  | ....$
2018.12.27 22:00:41 LOG4[27]: growing input buffer.
2018.12.27 22:00:41 LOG4[27]: Data received
2018.12.27 22:00:41 LOG4[27]: .49 9b 0e d4 5e f3 6f 63 c4 57 8c 9f a2 aa e3 02 | I...^.oc.W......
2018.12.27 22:00:41 LOG4[27]: .59 06 57 0c ed a7 55 bb 39 3f 41 1c 2c e2 78 18 | Y.W...U.9?A.,.x.
2018.12.27 22:00:41 LOG4[27]: .7b 82 f4 09                                     | {...
2018.12.27 22:00:41 LOG4[27]: stream ConstantCompare failed
2018.12.27 22:00:41 LOG4[27]: VerifyMac failed
2018.12.27 22:00:41 LOG3[27]: wolfSSL error occurred, error = 305 line:13107 file:src/internal.c
2018.12.27 22:00:41 LOG3[27]: wolfSSL error occurred, error = 312 line:9570 file:src/ssl.c
````

kaleb-himes, 2018-12-27:
Hi @roytam1,

I see you have the setting --enable-pkcs11 turned on. Are you using a hardware piece to offload AES operations perhaps? I ask because we had a very similar report in May of this year and it turned out to be a bug in the offload operations. A fix was posted here: https://github.com/wolfSSL/wolfssl/pull/1479

This was the error log from that report:

```
wolfSSL (1): got CHANGE CIPHER SPEC
wolfSSL (1): PadCheck failed
wolfSSL (1): VerifyMac failed
wolfSSL (0): wolfSSL error occurred, error = -305
wolfSSL (0): wolfSSL error occurred, error = -312
wolfSSL (2): wolfSSL Entering SSL_get_error
SSL Connection Negotiation Failed - Aborting
```

Let us know if you are using any hardware crypto solutions and if so could you temporarily try disabling the hardware crypto and test with just software to see if that resolves the issue?

Warm Regards, 

- K

roytam1, 2018-12-27:
and the problem is that, when comparing wolfssl's MAC verification function with mbedtls' one, lots of cases that exists in mbedtls are not in wolfssl.

roytam1, 2018-12-27:
and also, if client send Hello with `Extension: signature_algorithms`, and the cert's Signature Hash Algorithm Hash method doesn't match any of signature_algorithms, shouldn't wolfSSL downgrade TLS to 1.0 and continue?

kaleb-himes, 2019-01-22:
@roytam1,

wolfSSL will only downgrade if the correct method is used. For downgrade use wolfSSLv23_client_method. Also wolfSSL disables deprecated and unsafe protocols and ciphers by default. This is because we care very much about the safety of our users and try to protect them from accidentally using weak ciphers or protocols when avoidable. If you want to be able to downgrade to TLS v1.0 you will need to add the define ```WOLFSSL_ALLOW_TLSV10``` to your configuration as it is not available with default settings.

closed 2019-02-28 by kaleb-himes