wolfSSL/wolfssl #2002: Can't define WOLFSSL_PUB_PEM_TO_DER by build configuration

miluoalbert, 2018-12-27:
When I used API `wc_PemPubKeyToDer` in my program, I always got a error when linking libwolfssl.so:
```
xxxx/src/test.c:50: undefined reference to ‘wc_PemPubKeyToDer’
collect2: ld returned 1 exit status
```

I found this error is caused by lost of a define -- `WOLFSSL_PUB_PEM_TO_DER`, and I fixed it by add a line in my `include/wolfssl/wolfcrypt/settings.h`:
```
#define WOLFSSL_PUB_PEM_TO_DER
```

I just wondering if there is any way I can enable this definition by build configuration?

dgarske, 2018-12-27:
Hi miluoalbert,

Try using `./configure CFLAGS="-DWOLFSSL_PUB_PEM_TO_DER"`. Or its enabled when cert request extensions `WOLFSSL_CERT_EXT` are enabled, so you could also use `./configure --enable-certext`.

Thanks,
David Garske, wolfSSL

dgarske, 2019-01-17:
Hi miluoalbert,

I believe this issue has been resolved, so I am marking this as closed. Please let us know if you have any questions.

Thanks,
David Garske, wolfSSL

closed 2019-01-17 by dgarske