wolfSSL/wolfssl #2006: Warning although hardening options are defined

miluoalbert, 2018-12-28:
Get warning when link libwolfssll.so:
```
wolfssl/3.15.4/files/wolfssl/include/wolfssl/wolfcrypt/settings.h:1750:14: WARNING：#warning "For timing resistance / side-channel attack prevention consider using harden options" [-Wcpp]
             #warning "For timing resistance / side-channel attack prevention consider using harden options"
              ^
```

I think it is caused by #1016 `wolfssl/wolfcrypto/setting.h`:
https://github.com/wolfSSL/wolfssl/blob/c54e8e31bd3e55157f928bdf4917549ed6db9e8e/wolfssl/wolfcrypt/settings.h#L1762-L1775

Line1762 should be: 
```cpp
#ifdef WC_NO_HARDEN
```

dgarske, 2018-12-28:
Hi miluoalbert,

The line 1762 looks correct to me. Its a double negative. If you define `WC_NO_HARDEN` this section will be skipped. Otherwise its checking to make sure the timing resistance is enabled.

The most likely reason for this is your end application is not including `#include <wolfssl/options.h>` before any wolfSSL headers. The default `./configure` enables hardening and generates a build settings file at `wolfssl/options.h`. You application needs to also include this.

There is an open PR 
https://github.com/wolfSSL/wolfssl/pull/1899, which demonstrates a solution inside our library, although currently we recommend adding the wolfssl/options.h include in your end application.

Thanks,
David Garske, wolfSSL

dgarske, 2019-01-17:
Hi miluoalbert,

I believe this issue has been resolved, so I am marking this as closed. Please let us know if you have any questions.

Thanks,
David Garske, wolfSSL

closed 2019-01-17 by dgarske