wolfSSL/wolfssl #2039: Redundant Check in CheckBitString

ManSoSec, 2019-01-17:
I noticed a redundant check in CheckBitString:

https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L812
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L827

if ((idx + 1) > maxIdx)
        return BUFFER_E;

if (idx + 1 > maxIdx) {
        WOLFSSL_MSG("Attempted buffer read larger than input buffer");
        return BUFFER_E;
    }

The second one needs to be removed.

ejohnstown, 2019-01-17:
Thank you for bringing this to our attention. I'll go ahead and fix this and the other two issues.

ejohnstown, 2019-01-17:
I do not believe the second check is redundant. The variable `idx` is advanced in between the two checks.

ManSoSec, 2019-01-17:
Thank you for checking! The reason I reported that is because of in 2 other locations, there is only one check. So I thought those locations might need an additional check. Would you please check them:

https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L5448
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L2064

In both cases, GetLength is called in the same way in CheckBitString
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L818

ejohnstown, 2019-01-17:
`GetLength()` performs its own check on what it is going to read from the buffer against the buffer bounds. In `GetDateInfo()`, `GetASNObjectId()`, and `CheckBitString()`, we are first checking that the first byte, the ASN chunk type, is in the buffer. Then we call `GetLength()` which checks that the length is inside the buffer.

The checks aren't looking redundant to me.

ManSoSec, 2019-01-17:
ok, thanks for the information.

closed 2019-01-19 by ejohnstown