wolfSSL/wolfssl #2055: Port WolfSSL to OP-TEE

iovoi, 2019-01-23:
Hi, 
I am working to port the wolfssl to OP-TEE OS. OP-TEE does not follow the posix standards and it has its own socket interface, time interface, random number interface, single thread and no file system. I have compile wolfssl as a static library and implemented the above interfaces, xmalloc, xrealloc, xfree in wolfssl. I also have used the wolfssl_load_and verify_buffer or the like api handling the client certificate stuff. However, I can not make it connect to the wolfssl-example/server-tls-callback program. The server-tls-callback gives output as follows,

> Waiting for a connection...
> my_IORecv: received 5 bytes from 4
> my_IORecv: received 143 bytes from 4
> my_IOSend: sent 91 bytes to 4
> my_IOSend: sent 1201 bytes to 4
> my_IOSend: sent 338 bytes to 4
> my_IOSend: sent 9 bytes to 4
> Connection closed
> wolfSSL_accept error = -308

I can use the wolfssl-example/client-tls-callback to connect to server program which works perfectly. 
I traced all the way down to my client code in OP-TEE port. The state transitions for my client code in file src/ssl.c are: connect_begin, client_hello_sent, receives messages from server, hello_again, hello_again_reply, first_reply_done, first_reply_first,
then the program eventually comes to mp_read_radix in wolfcrypt/src/integer.c and this function returns error code -3 (MP_VAL) which means invalid input as it is defined as
> #define MP_VAL -3 /* invalid input */ 

So my questions are Why would this error occur? and How should fix this for it to work? 

The configuration I used for building wolfssl is:

> ./configure -prefix="/home/iovoi/projects/wolfssl" LIBS="-lc -lm -lgcc -l:libutee.a" --enable-examples=no --enable-crypttests=no --enable-valgrind=no --enable-iopool=no --enable-static --enable-singlethreaded=yes --host=arm-linux-gnueabihf CC=/home/iovoi/projects/op-tee-linux-qemu/toolchains/aarch32/bin/arm-linux-gnueabihf-gcc AR=/home/iovoi/projects/op-tee-linux-qemu/toolchains/aarch32/bin/arm-linux-gnueabihf-ar RANLIB=/home/iovoi/projects/op-tee-linux-qemu/toolchains/aarch32/bin/arm-linux-gnueabihf-ranlib --enable-filesystem=no CFLAGS="-DWOLFSSL_USER_IO -DNO_WRITEV -DWOLFSSL_DH_CONST -DNO_DEV_RANDOM -DUSER_TIME -DXMALLOC_USER -I/home/iovoi/projects/wolfssl/include" LDFLAGS="-L/home/iovoi/projects/wolfssl/lib"

I am new to wolfssl and not very familiar with tls connections. Help is needed here. 
Thank you so much in advance!

iovoi, 2019-01-23:
I finally resolved this, there are bugs in my code.

closed 2019-01-23 by iovoi

kaleb-himes, 2019-01-23:
Hi @iovoi,

I am sorry we missed the opportunity to assist you with this but we are glad to hear the issues were resolved! If you ever have any other questions please feel free to email us at support@wolfssl.com for best response times!

Can you tell us anything about what your project is and what you are working on? We are always eager to hear about the ways our customers are using our software to meet their needs!

Warm Regards,

- K

kaleb-himes, 2019-01-28:
Hi @iovoi,

Just reaching out one more time in case you missed my last. We are happy to support your current effort in any way we can, simply email us at support@wolfssl.com if you have any other questions come up. Also if you would like us to maintain your porting effort in all future releases of wolfSSL we are happy to accept any code contributions from you and/or your team, all we need is a contributor agreement to pull in any changes you wish to contribute to the master repository.

Let us know if we can help in any way.

Warm Regards,

- K

iovoi, 2019-05-21:
Hi Kaleb-himes,

Thank you for reaching out and sorry for the delay. 
I was working on a project to implement the client TA in OP-TEE which is an open-sourced secure OS running in TrustZone. I want to make it connect to a server in TLS. 
I have finished the prototype of the project and will email you if I have further questions.

Thank you very much!

Best,
iovoi