wolfSSL/wolfssl #2056: What is the likelihood of SHA-512/256 being implemented?

cryptocyclist, 2019-01-23:
(That is, SHA-512, but with different hash initialisation - as is done for SHA-384 - then truncated to 256 bits.)

With the same output length as SHA-256, this function is generally faster on modern (64-bit) hardware and not vulnerable to length extension attacks.

If not likely from the core developers, would a PR be accepted?  Would such as PR have to support SHA512/256 equivalents to WOLFSSL_ASYNC_MARKER_SHA384 and IntelQaSymSha384? (Hopefully not!)

And would "Sha512t256" be acceptable in the place of "Sha384" in function and macro names?

kaleb-himes, 2019-01-23:
Hi @cryptocyclist,

> (That is, SHA-512, but with different hash initialisation - as is done for SHA-384 - then truncated to 256 bits.)
> With the same output length as SHA-256, this function is generally faster on modern (64-bit) hardware and not vulnerable to length extension attacks.

It is not currently on our Road map.

> If not likely from the core developers, would a PR be accepted?

Thanks for reaching out. wolfSSL can accept Pull Requests if a contributor agreement is in place between you, your organization, and wolfSSL Inc. If that is something you would be interested in working on then feel free to open the PR and once it is ready shoot an email to support@wolfssl.com with the subject line: "Pertaining to PR #<number of PR here>, seeking contributor agreement" and we will send that document over for you to review.

> Would such as PR have to support SHA512/256 equivalents to WOLFSSL_ASYNC_MARKER_SHA384 and IntelQaSymSha384? (Hopefully not!)

The support would be up to you as the developer. If you open a PR we are happy to review and provide feedback on it.

> And would "Sha512t256" be acceptable in the place of "Sha384" in function and macro names?

No it would not be acceptable. It is (or should be) common knowledge to most cryptographers that SHA384 is really a truncated version of SHA512 but for non-crypto experts and industry standard guideline you will never see SHA384 referred to as "Truncated Sha512", it is always referred to as SHA384. If non-crypto experts want to review our code and see if we support SHA384, a function/macro name of SHA512t256 would not be relevant to them.

- K

cryptocyclist, 2019-01-23:
Regarding "Sha512t256", my question was about naming in a potential PR for SHA512/256, i.e. a PR that used a #define WOLFSSL_SHA512T256, a type wc_Sha512t256, and functions such as InitSha512t256(), wc_InitSha512t256(), wc_Sha512t256Update() and wc_Sha512t256Final().

Unwieldy, but probably the clearest...

kaleb-himes, 2019-01-23:
@cryptocyclist,

A that is much clearer, so sorry for the misunderstanding. Yes if we are talking about SHA512/256  and there is currently no other name then the variable and API choice of ```SHA512t256``` would be perfectly acceptable!

Is this for a commercial application or something you are working on as a hobby/side project?

- K

kaleb-himes, 2019-01-24:
Hi @cryptocyclist 

I came across something in one of our internal apps we use for testing that is relevant to your question. One of our developers left a comment:

```
/*TODO uncomment next cases if truncated hash algos get added to wolfCrypt
else if (strstr(line, ",SHA-512224]") != NULL)
tmp = CAVP_HASH_TYPE_SHA512224;
else if (strstr(line, ",SHA-512256]") != NULL)
tmp = CAVP_HASH_TYPE_SHA512256;
*/
```

So it looks like while not on the roadmap, someone has already considered the variable name: ```SHA-512256``` where you had asked about using ```SHA512t256```.

Regards,

- K