wolfSSL/wolfssl #2071: Wolfssl with haproxy

git001, 2019-02-02:
The issue #1335 is closed but I can't find a answer there if wolfssl is a possible TLS/SSL library for haproxy.

As I build haproxy with openssl and boringssl I'm courious if wolfssl is also a possible TLS/SSL library for haproxy, not only technically also from license point of view.

These are the image which I maintain.  
https://hub.docker.com/r/me2digital/haproxy19  
https://hub.docker.com/r/me2digital/haproxy-19-boringssl

dgarske, 2019-02-04:
Hi git001,

Yes we do have a port of haproxy for wolfSSL. Currently its in a private repo. Please email us directly at info@wolfssl.com and we should be able to get you a copy of the source code.

Thanks,
David Garske, wolfSSL

git001, 2019-02-05:
@dgarske thanks.

I have some questions before I try to create a new Docker image.

* Can I push a wolfSSL builded image on public docker hub?
* What's the license for the port of haproxy for wolfSSL?  
* Do you plan to provide the patches to the upstream?

kaleb-himes, 2019-02-28:
Being in Zendesk.

closed 2019-02-28 by kaleb-himes