wolfSSL/wolfssl #2079: Certificate verification failed for X25519 certificates

Ryetschye, 2019-02-08:
I will do a verifcation for a certificate chain but I receive an error:

ASN_UNKNOWN_OID_E

This will happen for certificates similar to the appended certificates:

[certificates.zip](https://github.com/wolfSSL/wolfssl/files/2844024/certificates.zip)

Looking to openSSL the verification process will be successful.
It seems that is the problem that wolfSSL does not accept a certificate where a X25519 key is used as a public key. The signing certificate holds an ED25519 to sign.

JacobBarthelmeh, 2019-03-01:
Thanks @Ryetschye for bringing this up. Parsing the X25519 OID from a certificate is not yet supported. We do though support certificates that have Ed25519 keys with Ed25519 signatures. Can you tell us some about your use case? In the mean time I'll add this to the desired feature request list.

Ryetschye, 2019-03-01:
The use-case is described easily when using Ed25519 certificates.
I want to generate a symmetric key for two entities and both entities have to authenticate each other.
With Ed25519 this would look like this:
1. generate a X255519 key pair
2. sign public key (with certificate from a PKI)
3. exchange public keys (including signature)
4. exchange signing certificates (these certificates might contain Ed25519 keys)
5. verify certificate + verify signature by using a stored sub root certificate
6. use X25519 keys for ECDH process

To make the whole stuff easier: put the public X25519 key into a certificate signed by a Ed25519 certificate from the PKI. For the receiver it is clear how to verify a certificate but it is not clear how to check the structure of a signed key (not well defined). Furthermore we use an HSM that creates X25519 certificates, but your wolfSSL based system cannot use it.
Last but not least we can add additional information into the certificate without adaption of any interface!

JacobBarthelmeh, 2019-03-05:
closing this issue, it is being handled through support@wolfssl.com

closed 2019-03-05 by JacobBarthelmeh