wolfSSL/wolfssl #2092: EVP_CipherFinal returns success (without updating destination length),  instead of error

vaintroub, 2019-02-14:
Below is an coding example of decryption that should fail, but it does not 
EVP_Cipher_Init, EVP_Cipher_Update,EV Cipher_Final all return success, although the last function
does not initialize destination length.

```
#include <openssl/evp.h>
#include <openssl/aes.h>
#include <openssl/err.h>

int main()
{
	const byte src[] = {'a'};

	byte key[] = "a\0\0\0\0\0\0\0\0\0\0\0\0\0\0";  
	byte *iv= 0;  /* align */

	byte out[AES_BLOCK_SIZE * 4];

	EVP_CIPHER_CTX de;
	int outlen ;
	int total = 0;

	EVP_CIPHER_CTX_init(&de);
	if (EVP_CipherInit(&de, EVP_aes_128_ecb(),
		(unsigned char*)key, (unsigned char*)iv, 0) != WOLFSSL_SUCCESS)
	{
		return -1;
	}
	
	if (EVP_CipherUpdate(&de, (byte*)out, &outlen,
				(byte*)src, 1) != WOLFSSL_SUCCESS)
	{
		return -1;
	}
	
	total += outlen;

	outlen = -1;
	if (EVP_CipherFinal(&de, (byte*)&out[total], &outlen) == WOLFSSL_SUCCESS)
	{
		printf("outlen = %d\n", outlen);
		/* BUG : EVP_CipherFinal completes successfully, without initializing output length, instead of failing with error */
		abort();
	}
	return 0;
}

kaleb-himes, 2019-02-14:
Hi @vaintroub,

Thank you for reaching out to wolfSSL about this issue. If at any time in the future you happen to experience delays when opening issues on github please feel free to contact support@wolfssl.com or open an issue through the zendesk portal at https://wolfssl.zendesk.com for faster response times.

I will attempt to build the above application and link it against wolfSSL's compatibility layer to reproduce. I will send you any notes soon.

Warm Regards,

K

JacobBarthelmeh, 2019-02-27:
Thank you for the report @vaintroub !
A fix and regression test have been submitted in this PR (https://github.com/wolfSSL/wolfssl/pull/2113)

closed 2019-02-28 by JacobBarthelmeh