wolfSSL/wolfssl #2097: couldn't convert EC public key to PEM with ECC_PUBLIC_KEYTYPE

DevNaga, 2019-02-15:
I have tried converting the EC pub / private keys to PEM. While the private key is successful, the pubkey fails with `BAD_FUNC_ARG`, when used `ECC_PUBLICKEY_TYPE`.

However, if i replace `ECC_PUBLIC_KEY` with `PUBLIC_KEY`, everything works fine and able to read them via `openssl ec -in` via command line. Is this the correct behavior or am i missing something totally?

Please see the below sample program,

```c
#include <stdio.h>
#include <stdint.h>

#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/settings.h>
#include <wolfssl/wolfcrypt/ecc.h>
#include <wolfssl/wolfcrypt/random.h>
#include <wolfssl/wolfcrypt/asn_public.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int main()
{
    uint8_t der_key[4096];
    uint8_t pem_out[4096];
    WC_RNG rng;
    ecc_key key;
    int ret;

    ret = wc_InitRng(&rng);
    if (ret != 0) {
        return -1;
    }

    ret = wc_ecc_init(&key);
    if (ret != 0) {
        return -1;
    }

    ret = wc_ecc_make_key_ex(&rng, 32, &key, ECC_SECP256K1);
    if (ret != 0) {
        return -1;
    }

    ecc_point pubkey;

    ret = wc_ecc_make_pub(&key, &pubkey);
    if (ret != 0) {
        return -1;
    }

    ret = wc_EccKeyToDer(&key, der_key, 4096);
    if (ret < 0) {
        return -1;
    }

    ret = wc_DerToPem(der_key, ret, pem_out, sizeof(pem_out), ECC_PRIVATEKEY_TYPE);
    if (ret < 0) {
        printf("cannot pem the privkey\n");
        return -1;
    }

    printf("generated ecc key\n");

    FILE *fp;

    // writing priv
    fp = fopen("./ecc_secp256k1.key", "w");
    if (!fp) {
        return -1;
    }

    fwrite(pem_out, ret, 1, fp);

    fclose(fp);

    // convert pub key to der
    ret = wc_EccPublicKeyToDer(&key, der_key, 4096, ECC_SECP256K1);
    if (ret < 0) {
        return -1;
    }

    // write der to pem
    memset(pem_out, 0, sizeof(pem_out));
    ret = wc_DerToPem(der_key, ret, pem_out, sizeof(pem_out), ECC_PUBLICKEY_TYPE);
    if (ret < 0) {
        printf("cannot pem the pubkey\n");
        return -1;
    }

    fp = fopen("./ecc_secp256k1_p.key", "w");
    if (!fp) {
        return -1;
    }

    fwrite(pem_out, ret, 1, fp);

    fclose(fp);

    return 0;
}
```

dgarske, 2019-02-20:
Hi DevNaga,

Thanks for the issue report and example code. I've put together a similar example for exporting an ECC private and public key here (https://github.com/wolfSSL/wolfssl-examples/pull/128). I also pushed a fix for `ECC_PUBLICKEY_TYPE` here (https://github.com/wolfSSL/wolfssl/pull/2106).

Please let me know if you have any further questions or issues.

Thanks,
David Garske, wolfSSL

closed 2019-02-20 by dgarske

DevNaga, 2019-02-20:
Hello David,

Thank you for the example and the fix.

Regards,
Dev

PR #2106 (2019-02-22): Adds `ECC_PUBLICKEY_TYPE` to the support PEM header types
Adds `ECC_PUBLICKEY_TYPE` to the support PEM header types. Fixes #2097.
commit d81fb727 (2019-02-22): Adds `ECC_PUBLICKEY_TYPE` to the support PEM header types. Fixes #2097.