wolfSSL/wolfssl #2126: stm32f4xx_cryp.h missing

Vaibhav1587, 2019-02-28:
Created project using STM32CubeMX. 
Trying to get the wolfssl test run on it. Referring youtube video [https://www.youtube.com/watch?v=FG6jvINvls0&t=507s](url)
Using TrueStudio 

Getting error,
Description	Resource	Path	Location	Type
fatal error: stm32f4xx_cryp.h: No such file or directory	settings.h	

copied missing files. 
Now getting following error ,

Description	Resource	Path	Location	Type
'wc_Sha256 {aka struct wc_Sha256}' has no member named 'buffer'	sha256.c	/STM32_SSH/Src/wolfssl/wolfcrypt/src	line 561	C/C++ Problem

 Is it really working ? Does anyone ever tried to run test ?

dgarske, 2019-02-28:
Hi Viabhav1587,

Those files are part of the STM32Cube project. The tool does not have a "checkbox" to indicate including the crypto or hash files, so you'll need to manually copy those over.

For example on my Mac they are installed into:
`/Users/[username]/STM32Cube/Repository/STM32Cube_FW_F4_V1.13.0/Drivers/STM32F4xx_HAL_Driver/Inc/`

`/Users/[username]/STM32Cube/Repository/STM32Cube_FW_F4_V1.13.0/Drivers/STM32F4xx_HAL_Driver/Src/`

The files that need copied are:

```
inc/stm32f4xx_hal_cryp.h
inc/stm32f4xx_hal_cryp_ex.h
inc/stm32f4xx_hal_hash.h
inc/stm32f4xx_hal_hash_ex.h
src/stm32f4xx_hal_cryp.c
src/stm32f4xx_hal_cryp_ex.c
src/stm32f4xx_hal_hash_ex.c
src/stm32f4xx_hal_hash.c
```

Additionally in your generated `stm32f4xx_hal_conf.h` file you must uncomment `#define HAL_HASH_MODULE_ENABLED` and `#define HAL_CRYP_MODULE_ENABLED`.

I've attached an example project.
[wolfSTM32_CubeMXF4.zip](https://github.com/wolfSSL/wolfssl/files/2916975/wolfSTM32_CubeMXF4.zip)

Let me know if you have any questions or issues.

Thanks,
David Garske, wolfSSL

Vaibhav1587, 2019-03-01:
In wolfssl/wolfcrypt/settings.h 

```
        #elif defined(WOLFSSL_STM32F4)
            #include "stm32f4xx.h"
            #ifdef STM32_CRYPTO
                #include "stm32f4xx_cryp.h"
            #endif
            #ifdef STM32_HASH
                #include "stm32f4xx_hash.h"
            #endif
```

should I rename the files to 

> stm32f4xx_hal_cryp.h
and 

> stm32f4xx_hal_hash.c

dgarske, 2019-03-02:
Hi Vaibhav1587,

Those lines are for using the STM32 Standard Peripheral Library, which you are not using.

For the STM32 CubeMX you must define `WOLFSSL_STM32_CUBEMX` and it will include `WOLFSSL_STM32_CUBEMX` settings.h:1120.

Thanks,
David Garske, wolfSSL

dgarske, 2019-03-14:
Hi Vaibhav1587,

I believe this issue is resolved, so closing it. If you have any updates to add to this issue that would be helpful.

Thanks,
David Garske, wolfSSL

closed 2019-03-14 by dgarske

seemakumar8, 2020-05-06:
> Hi Viabhav1587,
>
> Those files are part of the STM32Cube project. The tool does not have a "checkbox" to indicate including the crypto or hash files, so you'll need to manually copy those over.
>
> For [...]

Hi, can you also please share wolfssl folder you use for the example which is present as a softlink under Middlewares/Third_Party folder. I tried with downloading it from GitHub and excluding some files from build. I can compile successfully, but cannot run. There is a hardware fault at the beginning itself. 

Thanks

dgarske, 2020-05-11:
Hi Seemakumar8,

I used the wolfSSL v4.3.0 release. Make sure you increase the stack and heap space. If you'd like support please email us at support@wolfssl.com <mailto:support@wolfssl.com> and include details about how your are building, what the target is and what error(s) you are seeing.

Thanks,
David Garske
530-409-2990
skype: davidgarske
www.wolfssl.com <http://www.wolfssl.com/>
https://github.com/wolfssl <https://github.com/wolfssl>
http://twitter.com/wolfSSL <http://twitter.com/wolfSSL>

wolfSSL has TLS1.3! https://wolfssl.com/tls13 <https://wolfssl.com/tls13>