wolfSSL/wolfssl #2132: Build broken with busybox shell

mwarning, 2019-03-01:
wolfssl fails to compile in docker alpine containers because the shell there is the ash shell (busybox):

```
configure: WARNING: unrecognized options: --disable-nls
configure: loading site script /home/user/build/openwrt/include/site/mips
./configure: line 2: syntax error: unterminated quoted string
make[3]: *** [Makefile:178: /home/user/build/openwrt/build_dir/target-mips_24kc_musl/wolfssl-3.15.3-stable/.configured_b7642ce15094029926a9913be4cb199e] Error 2
make[3]: Leaving directory '/home/user/build/openwrt/package/libs/wolfssl'
time: package/libs/wolfssl/compile#12.30#0.59#14.02
make[2]: *** [package/Makefile:109: package/libs/wolfssl/compile] Error 2
make[2]: Leaving directory '/home/user/build/openwrt'
make[1]: *** [package/Makefile:103: /home/user/build/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/build/openwrt'
make: *** [/home/user/build/openwrt/include/toplevel.mk:218: world] Error 2
```

The generated configure script starts with `#! /bin/sh` which seem to cause the failure.

The cause of the problem is this commit: https://github.com/wolfSSL/wolfssl/commit/1b9cff1c5d09adb0af858992e2a2df01b1d4dc58 (found and confirmed via git bisect and `./autogen.sh && ./configure`)

mwarning, 2019-03-01:
Context: https://forum.openwrt.org/t/cmake-build-error/25140

ejohnstown, 2019-03-04:
Please see PR #2137 for the fix.

mwarning, 2019-03-05:
@ejohnstown thanks, that would definitely "fix" the issue. But it is not clear if this is a problem with libtool or some other component.

mwarning, 2019-03-05:
Here are the steps to reproduce this error:

File `Dockerfile`:
```
FROM alpine:3.9

RUN apk add git build-base bash ncurses-dev perl tar findutils patch coreutils gawk grep unzip bzip2 wget python2-dev curl diffutils bsd-compat-headers less diffutils zlib-dev file sudo autoconf automake libtool && \
    addgroup -S user && \
    adduser -D -G user user && \
    echo 'user ALL=NOPASSWD: ALL' > /etc/sudoers.d/user

USER user
WORKDIR /home/user
```
Create Docker image:
```
docker build -t wolfssl_alpine .
```
Run container:
```
mkdir ~/mybuild
docker run -v ~/mybuild:/home/user -it wolfssl_alpine /bin/bash
```

Compile wolfssl:
```
wget https://github.com/wolfSSL/wolfssl/archive/v3.15.7-stable.tar.gz
tar -xvf v3.15.7-stable.tar.gz
cd wolfssl-3.15.7-stable/
./autogen.sh
./configure
```

ejohnstown, 2019-03-05:
Thank you for the bug report and description. I have a new pull request with the fix. See PR #2139.

closed 2019-03-05 by ejohnstown