wolfSSL/wolfssl #2150: DTLS v1.3 support

DurandA, 2019-03-11:
The readme states that DTLS 1.3 is supported:
> wolfSSL supports industry standards up to the current TLS 1.3 and DTLS 1.3 levels

However, I cannot find any reference to `wolfDTLSv1_3` in *ssl.h*. Is this feature available somewhere on an development branch? I know that DTLS 1.3 is still in draft so I expected support for a recent draft version.

ejohnstown, 2019-03-11:
That is a typing mistake. I have fixed it with PR #2152.

ejohnstown, 2019-03-11:
It looks like they are still grinding away at the RFC draft. We are doing initial investigation on the changes needed to our DTLS implementation to work with v1.3.

Thank you for your interested, and thank you for the bug report.

closed 2019-03-11 by ejohnstown

rizlik, 2022-06-22:
Hi @DurandA,
just let you know that wolfSSL now supports DTLSv1.3 from commit `fb704774a0e961685e1c082ea13b3e71a4c0d953` !
Marco