wolfSSL/wolfssl #2155: Missing documentation of cipher names

DurandA, 2019-03-12:
When specifying cipher list, it is cumbersome to find cipher names in wolfSSL format (e.g. `TLS13-AES128-CCM-8-SHA256`):
https://github.com/wolfSSL/wolfssl/blob/6e1b05316d9b05ff86513d3d77b1ba7834c320d1/wolfssl/ssl.h#L714-L716

In the manual, [TLS cipher names are used](https://www.wolfssl.com/docs/wolfssl-manual/ch4/#h.9d0g1txryomj) and we have to crawl test configurations (e.g. [test.conf](https://github.com/wolfSSL/wolfssl/blob/6e1b05316d9b05ff86513d3d77b1ba7834c320d1/tests/test.conf)) to find them.

dgarske, 2019-03-12:
Hi DurandA,

Thanks for your feedback on the user manual. We will work on improving that table of cipher suites.

For reference the complete cipher suite table with internal name and IANA name is here:
https://github.com/wolfSSL/wolfssl/blob/master/src/internal.c#L15968

You can also get a list of supported cipher suites at run-time using: 

```
void ShowCiphers(void)
{
    char ciphers[4096];
    int ret = wolfSSL_get_ciphers(ciphers, (int)sizeof(ciphers));
    if (ret == WOLFSSL_SUCCESS)
        printf("%s\n", ciphers);
}
```

Thanks,
David Garske, wolfSSL

DurandA, 2019-03-12:
Excellent, thanks for the tip!

closed 2019-03-12 by DurandA

DurandA, 2019-03-12:
Interestingly, GitHub code search is limited to small files:
> Only files smaller than 384 KB are searchable.

For larger files, search is only performed on the first 384 KB and *internal.c* was a bit too big. :laughing: