wolfSSL/wolfssl #2167: Compiling with CryptoAuthLib fails

DurandA, 2019-03-18:
Compiling *wolfSSL* with *CryptoAuthLib* triggers an ``undefined reference to `atcab_printbin_label'``.

Here is a more complete trace:
```
  CCLD     wolfcrypt/benchmark/benchmark
src/.libs/libwolfssl.so: undefined reference to `atcab_printbin_label'
collect2: error: ld returned 1 exit status
Makefile:3395: recipe for target 'wolfcrypt/benchmark/benchmark' failed
```

I configured *wolfSSL* to use *CryptoAuthLib* using `./configure --with-cryptoauthlib="/home/pi/cryptoauthlib/"`. *CryptoAuthLib* was installed from latest master:
```
git clone https://github.com/MicrochipTech/cryptoauthlib.git && cd cryptoauthlib/lib
cmake . -DATCA_HAL_I2C=ON -DATCA_BUILD_SHARED_LIBS=ON
make
```

dgarske, 2019-03-18:
Hi DurandA,

Thanks for your question. It looks like we made a local change to our fork of CryptoAuthLib that was not accounted for:
https://github.com/dgarske/cryptoauthlib/blob/atcatls/lib/basic/atca_helpers.h#L49

Specifically we added a macro to stub 

We are in a release cycle, but after it is complete I will change wolfcrypt/port/atmel/atmel.c to the following at line 108:

```
#ifdef ATCAPRINTF
atcab_printbin_label((const char*)"\r\nRandom Number", rand_out, count);
#endif
```

dgarske, 2019-03-18:
Hi DurandA,

Sorry about sending too soon and accidentally closing. Updated the note above.

Thanks,
David Garske, wolfSSL

DurandA, 2019-03-18:
I confirm that the above modification solves this issue. However a new `undefined reference` is thrown later:
```
  CCLD     tests/unit.test
tests/tests_unit_test-api.o: In function `test_wc_ecc_verify_hash_ex':
api.c:(.text+0x8938): undefined reference to `wc_ecc_sign_hash_ex'
api.c:(.text+0x8a20): undefined reference to `wc_ecc_sign_hash_ex'
api.c:(.text+0x8a40): undefined reference to `wc_ecc_sign_hash_ex'
api.c:(.text+0x8a60): undefined reference to `wc_ecc_sign_hash_ex'
api.c:(.text+0x8a80): undefined reference to `wc_ecc_sign_hash_ex'
tests/tests_unit_test-api.o:api.c:(.text+0x8aa4): more undefined references to `wc_ecc_sign_hash_ex' follow
```
This happens because [wcc_ecc_sign_hash_ex is not implemented if WOLFSSL_ATECC508A is defined](https://github.com/wolfSSL/wolfssl/blob/0d48a4a1a77126519cb1bd71959e3dfe249165be/wolfcrypt/src/ecc.c#L4441-L4452).

dgarske, 2019-03-18:
Hi DurandA,

Thanks for this report. Looks like you have `WOLFSSL_PUBLIC_MP` defined. I'm guessing you also have the `--enable-opensslextra` option set? I've pushed fixes for this to a branch here (https://github.com/dgarske/wolfssl/tree/atecc) and will push as a PR once the release is done.

[atecc.diff.txt](https://github.com/wolfSSL/wolfssl/files/2979739/atecc.diff.txt)

David Garske, wolfSSL

DurandA, 2019-03-18:
Thank you very much for investigating in this issue. I do have `WOLFSSL_PUBLIC_MP` but no `--enable-opensslextra` option. I will try your branch and report back.

DurandA, 2019-03-19:
It compiles fine from your branch! Thank you very much.

As a last step, I had to manually include *atmel.h*. It would be nice to do it automatically on `make install` (e.g. copy to */usr/local/include/wolfssl/wolfcrypt/port/atmel/atmel.h*) when configured with the `--with-cryptoauthlib` switch.

I will let you close the issue in case you want to keep it open until your branch is merged.

PR #2172 (2019-03-28): Fixes for ATECC support
* Fix for building ATECC without `ATCAPRINTF` defined.
* Fix for build warning with x/y always true when not building with ALT_ECC_SIZE.
* Fix for build error with undefined `wc_ecc_sign_hash_ex` when building ATECC and `WOLFSSL_PUBLIC_MP`.
* Fix to include the atmel.h with make install when using `--with-cryptoauthlib`.

Fixes #2167.
closed 2019-03-28 by ejohnstown