wolfSSL/wolfssl #2169: mp_read_unsigned_bin throw realloc(): invalid pointer on armv6l

DurandA, 2019-03-19:
Using `mp_read_unsigned_bin()` fails on *armv6l GNU/Linux* (a Raspberry Pi running Raspbian):
```
*** Error in `./test': realloc(): invalid pointer: 0xb6fd7fc4 ***
Aborted
```

Note that it works fine on *x86_64*.

Here is a minimal example to reproduce the issue:
```
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/ecc.h>

int main(void){
  char signature[64] = "";
  mp_int r, s;
  mp_read_unsigned_bin(&r, signature, 32);
  mp_read_unsigned_bin(&s, signature+32, 32);
  return 0;
}
```

It was compiled using `gcc -Wall -o test test.c -lwolfssl`. The output of `./configure` can be found [here](https://pastebin.com/0MZvLVPB).

kaleb-himes, 2019-03-19:
Hi @DurandA,

Could you trying calling ```wolfSSL_Init()``` before calling the mp functions and let me know if this resolves the issue.

Warm Regards,

K

DurandA, 2019-03-21:
Using `mp_init(&r); mp_init(&s);` solved the issue, thanks. I was not aware that these should be initialized.

I extract *x* and *y* components of public keys and *r* and *s* components of signatures to create [COSE](https://tools.ietf.org/html/rfc8152) keys and signatures. These are using raw points instead of DER/X9.63. Is there a better way to do it?

closed 2019-03-21 by DurandA

kaleb-himes, 2019-03-25:
@DurandA,

I believe the best way to do this would be with the ```wc_ecc_sig_to_rs``` API which takes in a signature and outputs the raw r & s to the supplied buffers.

Warm Regards,

K

DurandA, 2019-04-02:
Thank you very much. It would be nice to update the [Doxygen](https://www.wolfssl.com/doxygen/group__ECC.html) so your users don't miss it. :smiley: