wolfSSL/wolfssl #2171: Allow use in GPLv3 projects

shiftee, 2019-03-21:
Hi,

I prefer to use the GPLv3 and will likely want to add encryption to a project i'm working on at the moment.

You only list the GPLv2 as your license which doesn't work for me as it is incompatible with the GPLv3.

Would it be possible to change to either **GPLv2+** or **GPLv2+GPLv3**?

Thanks

kaleb-himes, 2019-03-21:
Hi @shiftee,

If you send us an email at support@wolfssl.com with your contact information I can put you in touch with the right resource on our end to discuss this with you further. You can also contact our main office by the phone number listed on https://www.wolfssl.com/contact/ at your convenience if you prefer.

Warmest Regards,

K

closed 2019-03-21 by kaleb-himes

spinda, 2019-05-12:
Hi @kaleb-himes,

I think this is a rather important oversight. For example, your own recently-released [tiny-curl](https://daniel.haxx.se/blog/2019/05/11/tiny-curl/), designed for use with wolfSSL, would actually appear to be legally incompatible, as tiny-curl is made available under GPLv3 vs wolfSSL's GPLv2.

This is why the FSF's recommended  disclaimer stipulates something along the lines of "either version 2 of the License, or (at your option) any later version".

kaleb-himes, 2019-05-13:
Hi @spinda,

Thanks so much for your concern. We do use the noted disclaimer in every single source and header file. REF: https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/internal.h#L9-L10

As I mentioned to @shiftee any licensing concerns can readily be addressed by our sales and management staff.

Here on github you will only get responses from our engineers who do not deal with the legal side of our business much and are not prepared to distribute legal advice or input accordingly. If you have concerns please do not hesitate to open a support issue which can then be assigned to one of our Business associates for answering legal questions and / or our management team can be reached at the contact number on our website: https://www.wolfssl.com/contact/

Warmest Regards,

K