wolfSSL/wolfssl #2188: "host name validation"

YYangJ, 2019-04-01:
wolfSSL_X509_check_host() function is defined but not referred,does the latest version support "host name validation"?

kaleb-himes, 2019-04-04:
@YYangJ,

Yes wolfSSL supports this, did you configure with --enable-opensslall to get access to the API in src/ssl.c?

- KH

YYangJ, 2019-04-15:
No,How can I enable this feature in Windows environment?I try to call the function in ProcessPeerCerts, and it seems to work well,does wolfssl team plan to add macro to enable/disable "host name validation"?

dgarske, 2019-04-15:
Hi YYangJ,

Have you tried using our `wolfSSL_check_domain_name(ssl, domain);` API? This should be called before `wolfSSL_connect` and sets a hostname/domain name, that will be checked on connect.

Failures will be reported like this:

```
ret = wolfSSL_connect(ssl);
if (ret != WOLFSSL_SUCCESS) {
    err = wolfSSL_get_error(ssl, 0);
}
```

The `err` will be `DOMAIN_NAME_MISMATCH`.

Example:
https://github.com/wolfSSL/wolfssl/blob/master/examples/client/client.c#L2761

Thanks,
David Garske, wolfSSL

YYangJ, 2019-04-16:
Hi David Garske,
  I will try to use wolfSSL_check_domain_name(ssl, domain);API and give you feedfack ASAP.Thanks for your prompt and patient answers.

YYangJ, 2019-04-17:
Hi David Garske,
I used wolfSSL_check_domain_name(ssl, domain);API before wolfssl_connect,ProcessPeerCerts() function will check whether domain name match common name and alt names,I added wolfSSL_X509_check_host in ProcessPeercerts before, it looks like the same effect as calling wolfSSL_check_domain_name() before wolfssl_connect.

Thanks

YYangJ, 2019-04-17:
Hi David Garske,
I have another question,if the host is IP address, using wolfSSL_check_domain_name(ssl, domain);API before wolfssl_connect，will the SSL connection be established successfully or not? 

Thanks

dgarske, 2019-04-22:
Hi YYangJ,

If the domain name check fails then `wolfSSL_Connect(ssl)` will fail with `WOLFSSL_FAILURE`. You can see the error code using `wolfSSL_get_error(ssl, 0)`.

Thanks,
David Garske, wolfSSL

dgarske, 2019-05-07:
Hi YYangJ,

The `domain` argument is a string that is matched again subject common name or alternate names from the peer certificate. It does work with IP addresses if they are in a string format "192.168.0.1". Note: its very uncommon to use an IP address in a certificate, but is allowed. The certificate fields allow wildcard strings `*`. You can see the code in `MatchDomainName` here: https://github.com/wolfSSL/wolfssl/blob/683646e4526326e5ab757949249e04c21f241a13/src/internal.c#L8102

Thanks,
David Garske, wolfSSL

kaleb-himes, 2019-05-15:
Just to come back to your earlier question:

>> Yes wolfSSL supports this, did you configure with --enable-opensslall to get access to the API in src/ssl.c?

> No,How can I enable this feature in Windows environment?I try to call the function in ProcessPeerCerts, and it seems to work well,does wolfssl team plan to add macro to enable/disable "host name validation"?

To enable opensslall in windows add this define to wolfssl-x.x.x/IDE/WIN/user_settings.h:

```
#define OPENSSL_ALL
```

- K

closed 2020-01-23 by kaleb-himes