wolfSSL/wolfssl #2214: Failure in "make check" unit test: "double free or corruption"

RMartinDielectron, 2019-04-23:
[test-suite.log](https://github.com/wolfSSL/wolfssl/files/3105736/test-suite.log)
Memory freeing error occurs in unit test during "make check".   The error occurs using the zip files on the website and doing a clone from github.   The test-suite.log file is attached.

Compilation done using Linux Mint distribution 18.2.

RMartinDielectron, 2019-04-23:
On a hunch, I moved --enable-all to the end of the configure run and rebuilt.   With no other changes, it was able to run the unit tests.

embhorn, 2019-04-23:
Hello @RMartinDielectron 

It sounds like you were able to resolve the issue by changing the configuration. Are you able to reproduce the original issue? If so could you please share the configuration settings you used?

Best regards,

Eric Blankenhorn
wolfSSL Support

embhorn, 2019-04-30:
Hello @RMartinDielectron

Since we never heard back from you I will go ahead and close out this support inquiry. Should any other questions come up please do not hesitate to open a new issue anytime by emailing support@wolfssl.com or through the zendesk portal at https://wolfssl.zendesk.com. We are always happy to help out in any way we can.

Thank you for using wolfSSL support. We look forward to hearing from you in the future if you have any other questions.

Warmest regards,

Eric Blankenhorn
wolfSSL Support

closed 2019-04-30 by embhorn

RMartinDielectron, 2019-04-30:
My apologies.  I have been trying to systematically reproduce what was
happening but on a clean Ubuntu system.

Rob Martin
RMartin@dielectronsecurity.com
602-738-8010