wolfSSL/wolfssl #2233: Unable to set cipher list

efdalustaoglu, 2019-05-09:
I'm setting the cipher list as follows
```
const char *CIPHER_LIST = "PSK-AES128-CCM-8";
WOLFSSL_CTX *ctx = wolfSSL_CTX_new(wolfDTLSv1_2_client_method());
wolfSSL_CTX_set_cipher_list(ctx, CIPHER_LIST);

WOLFSSL *ssl = wolfSSL_new(ctx)
wolfSSL_set_cipher_list(ssl, CIPHER_LIST);
```
However when I print them afterwards, the result will be like

```
DHE-RSA-AES128-SHA
DHE-RSA-AES256-SHA
DHE-PSK-AES256-GCM-SHA384
DHE-PSK-AES128-GCM-SHA256
PSK-AES256-GCM-SHA384
PSK-AES128-GCM-SHA256
DHE-PSK-AES256-CBC-SHA384
DHE-PSK-AES128-CBC-SHA256
PSK-AES256-CBC-SHA384
PSK-AES128-CBC-SHA256
PSK-AES128-CBC-SHA
PSK-AES256-CBC-SHA
DHE-PSK-AES128-CCM
DHE-PSK-AES256-CCM
PSK-AES128-CCM
PSK-AES256-CCM
PSK-AES128-CCM-8
PSK-AES256-CCM-8
DHE-RSA-AES128-SHA256
DHE-RSA-AES256-SHA256
DHE-RSA-AES128-GCM-SHA256
DHE-RSA-AES256-GCM-SHA384
EDH-RSA-DES-CBC3-SHA
```
Also if I call `wolfSSL_get_cipher(ssl)` I got result "NONE". Is there anything that I'm doing wrong? I would like to enable only `PSK-AES128-CCM-8` encryption method. Is it possible?

efdalustaoglu, 2019-05-10:
Thanks for quick response,

I thought setting the cipher list `wolfSSL_set_cipher_list(ssl, CIPHER_LIST)`  will update the list that is coming from `wolfSSL_get_cipher_list(i)`, however it was instead the current available cipher list which I didn't know. And since I cannot make a successful handshake with the server at the moment the result from `wolfSSL_get_cipher(ssl)` is "NONE". Thanks again for the information.

closed 2019-05-10 by efdalustaoglu

CMCDragonkai, 2020-01-22:
I'm using curl built with wolfssl. And I also get this problem where I try to set `--ciphers PSK-AES256-CBC-SHA` even though it's listed in: https://curl.haxx.se/docs/ssl-ciphers.html

```
curl 7.67.0 (x86_64-pc-linux-gnu) libcurl/7.67.0 wolfSSL/4.3.0 zlib/1.2.11 libssh2/1.9.0 nghttp2/1.40.0Release-Date: 2019-11-06
Protocols: dict file ftp ftps gopher http https imap imaps pop3 pop3s rtsp scp sftp smtp smtps telnet tftp 
Features: AsynchDNS GSS-API HTTP2 IPv6 Kerberos Largefile libz SPNEGO SSL UnixSockets
```