wolfSSL/wolfssl #2249: MP_WARRAY with NO_64BIT causing shift-count-overflow

danielinux, 2019-05-20:
When compiled in 32bit mode with NO_64BIT, gcc produces many `shift-count-overflow` warnings for each MP_WARRAY usage:

```
./wolfssl/wolfcrypt/integer.h:204:32: warning: left shift count >= width of type [-Wshift-count-overflow]
 #define MP_WARRAY  ((mp_word)1 << (sizeof(mp_word) * CHAR_BIT - 2 * DIGIT_BIT + 1))
```

Reproducible with:

```
gcc -m32 -c -Wall -Wextra -std=c99 wolfcrypt/src/integer.c -I. -DNO_64BIT
```

kaleb-himes, 2019-05-21:
I'll look into it, thanks Daniele!

kaleb-himes, 2019-06-11:
Can reproduce with: ```./configure --build=i686-pc-linux-gnu "CFLAGS=-m32 -std=c99 -DNO_64BIT" "LDFLAGS=-m32" --disable-sha512 --disable-sha384 && make```

This build is currently supported: ```./configure --build=i686-pc-linux-gnu "CFLAGS=-m32 -std=c99 -DNO_64BIT" "LDFLAGS=-m32" --disable-sha512 --disable-sha384 --enable-fastmath --disable-asm && make```

Will using the fastmath library work for now?

- K

kaleb-himes, 2019-06-11:
Daniele,

Found the areas that would need resolved namely:

wolfssl/wolfcrypt/integer.h MP_WARRAY declaration would need to either adjust for the sizeof(mp_word) reducing from 8 to 4 in the 32-bit case with NO_64BIT set and each of the below calculations similarly would need adjusted based on change in sizeof(mp_word) which is only 4 in the 32-bit case with NO_64BIT defined.

```
wolfcrypt/src/integer.c: In function ‘mp_exptmod_fast’:
wolfcrypt/src/integer.c:1971:24: error: left shift count >= width of type [-Werror=shift-count-overflow]
           P->used < (1 << ((CHAR_BIT * sizeof (mp_word)) - (2 * DIGIT_BIT)))) {
                        ^~
wolfcrypt/src/integer.c: In function ‘mp_montgomery_reduce’:
wolfcrypt/src/integer.c:2417:10: error: left shift count >= width of type [-Werror=shift-count-overflow]
       (1 << ((CHAR_BIT * sizeof (mp_word)) - (2 * DIGIT_BIT)))) {
          ^~
wolfcrypt/src/integer.c: In function ‘mp_sqr’:
wolfcrypt/src/integer.c:2819:13: error: left shift count >= width of type [-Werror=shift-count-overflow]
          (1 << (sizeof(mp_word) * CHAR_BIT - 2*DIGIT_BIT - 1))) {
             ^~
wolfcrypt/src/integer.c: In function ‘mp_mul’:
wolfcrypt/src/integer.c:2856:12: error: left shift count >= width of type [-Werror=shift-count-overflow]
         (1 << ((CHAR_BIT * sizeof (mp_word)) - (2 * DIGIT_BIT)))) {
            ^~
  CC       tests/tests_unit_test-srp.o
wolfcrypt/src/integer.c: In function ‘s_mp_mul_digs’:
wolfcrypt/src/integer.c:3319:14: error: left shift count >= width of type [-Werror=shift-count-overflow]
           (1 << ((CHAR_BIT * sizeof (mp_word)) - (2 * DIGIT_BIT)))) {
              ^~
wolfcrypt/src/integer.c: In function ‘s_mp_mul_high_digs’:
wolfcrypt/src/integer.c:3827:10: error: left shift count >= width of type [-Werror=shift-count-overflow]
       (1 << ((CHAR_BIT * sizeof (mp_word)) - (2 * DIGIT_BIT)))) {
          ^~

```

That being said does it really make sense to be setting NO_64BIT on a 32-bit system? Have you come across some 32-bit architecture that really does not have a 64-bit types available?

danielinux, 2019-06-17:
@kaleb-himes you are right, I've checked and NO_64BIT seems to be safely turned off on those targets. I've always assumed that this define prevented operations that are not implemented even if the type is defined (e.g. `udiv64`). Feel free to ignore, perhaps all we need is a comment in the docs (or removing NO_64BIT altogether define if it leads to broken builds)

kaleb-himes, 2019-06-17:
Ahhh gotcha, Ya maybe we should try to make it more clear that the setting NO_64BIT means 64-bit arithmetic can not be performed on this system. I did have to use this setting on a 16-bit architecture once where the supported types were: 8-bit, 16-bit, and 32-bit.
There was no 64-bit type available so the setting was applicable in that case.

- KH

closed 2019-12-18 by kaleb-himes