wolfSSL/wolfssl #2263: incorrect input argument check in wolfSSL_EVP_CipherFinal

vaintroub, 2019-06-02:
Commit https://github.com/wolfSSL/wolfssl/commit/1605ab86d28340c30098bea9dce77756fddaab70

adds an incorrect parameter check for the value of `*outl` argument for the function `wolfSSL_EVP_CipherFinal()`. Now it reads as

    if (ctx == NULL || out == NULL || outl == NULL || (*outl < 0))
        return BAD_FUNC_ARG;

However, `outl` is an **output** parameter , not input, and it does not have to be initialized at all, and definitely does not have to be >=0 . pointer needs to be valid, the value should be set by the function. not checked by the function.

kaleb-himes, 2019-06-05:
@vaintroub,

Thank you for your report. A proposed solution has been putup in PR #2268 

Warm Regards,

KH

kaleb-himes, 2019-06-11:
Solution is merged. Please let us know if you have any other issues.

Warm Regards,

K

closed 2019-06-11 by kaleb-himes