wolfSSL/wolfssl #2264: EVP_CipherUpdate can read memory out-of-bounds

vaintroub, 2019-06-02:
This line https://github.com/wolfSSL/wolfssl/blob/10dde24363810b97ef83492f9c550dfc2a827995/wolfcrypt/src/evp.c#L361

can access up to 16 bytes of memory out-of-bounds. The `out` buffer in this case (decryption, no-padding) can/should be as big as input, but in some circumstances the function will try to copy 16 bytes beyond the end of it.

It is also good question why EVP_CipherUpdate would try to save the "last block" , since this lastBlock seems to be of no use EVP_CipherFinal in special case of no-padding.

vaintroub, 2019-06-02:
MariaDB ran into this bug here https://jira.mariadb.org/browse/MDEV-19581

kaleb-himes, 2019-06-05:
Under Review.

kaleb-himes, 2019-06-06:
Test case added to detect moving forward and fix supplied in #2272 

Thank you so much for the report @vaintroub, let us know if you find anything else. For valid bug reports our HR department is more than happy to put together a gift basket for your trouble, if you want to receive some "swag" (mugs, stickers, other logo items that are import friendly even overseas) please shoot me an email directly at kaleb@wolfssl.com with a physical mailing address and I will be happy to have our HR department send you a gift basket for your efforts!

Warm Regards,

K

kaleb-himes, 2019-06-11:
Solution has been merged. Please let us know if you have any other issues.

Warm Regards,

K

closed 2019-06-11 by kaleb-himes