wolfSSL/wolfssl #2330: DTLS server: non-conforming ServerHello response to early ChangeCipherSpec

pfg666, 2019-07-04:
Hi,

My research group has been doing analysis of DTLS implementations using state fuzzing/model learning. We also analyzed wolfssl via the examples/server/server application. We found that wolfssl responds to an unexpected ChangeCipherSpec by re-transmitting ServerHello. This happens if the ChangeCipherSpec is sent before it should but after the ClientHello-ServerHello exchange.

From my understanding of the DTLS 1.2 specification, an adequate response would have been either an UnexpectedMessage  Alert  or no response at all (silent discard). The latter is how wolfssl responds to other misplaced handshake messages, which I think is the best way to do so.

I attached a [.zip archive](https://github.com/wolfSSL/wolfssl/files/3360169/bad_serverhello.zip) containing the capture of a partial rsa handshake exposing this along with the application's output. Note that this occurs also if other key exchange algorithms are used (ecdhe, psk...).  I am also posting a screenshot of the capture so the scenario is clear.

[image]

It is a small issue unlikely to cause harm but I thought it was still worth reporting.

Cheers, Paul.

ejohnstown, 2019-09-23:
Thank you for the notifying us about this. I'll be taking a look at it this week. I may have some questions later.

ic0ns, 2019-09-30:
Hey,
we analyzed the problem further and it seems that this is intentional by your library.
From "internal.c":

> /**
             * on server side, retransmission is being triggered only by sending
             * first message of given flight, in order to trigger client
             * to retransmit its whole flight. Sending the whole previous flight
             * could lead to retransmission of previous client flight for each
             * server message from previous flight. Therefore one message should
             * be enough to do the trick.
             */

This results in the Server not transmitting the whole flight, but only the ServerHello. We understand the intention, but this behavior is kind of weird and unique for WolfSSL. We recommend sticking to the specification here and transmitting the whole flight. Additionally, a retransmission seems to be the wrong response, since the sequence number should give enough information to indicate that the message was out of place.

ejohnstown, 2019-12-31:
I have a pending fix for the issue @pfg666 reported. I agree that it should be taken care of. For the issue @ic0ns reported, I'm still thinking about a solution.

Thank you.

ejohnstown, 2021-06-17:
@ic0ns, the issue is fixed now. It has been for a while now. My colleague changed the retransmit back to the method described in the RFC. (The abbreviated version is an option now.) Also, with the other fix, the flight isn't retransmit with the out of place CCS message, the CCS message out of place is ignored.

closed 2021-06-17 by ejohnstown