wolfSSL/wolfssl #2342: Uninitialized read in function fp_leading_bit, for example server

maroneze, 2019-07-11:
I am trying to analyze WolfSSL (master at commit 651087ab5a1) with Frama-C, a code analysis tool, and I got a report of an uninitialized read in function `fp_leading_bit`, in expression `if (a->used != 0)`, for the following callstack:

```
fp_leading_bit (wolfcrypt/src/tfm.c:2753) ←
mp_leading_bit (wolfcrypt/src/tfm.c:3239) ←
StoreECC_DSA_Sig (wolfcrypt/src/asn.c:13358) ←
wc_ecc_sign_hash (wolfcrypt/src/ecc.c:4528) ←
EccSign (src/internal.c:3593) ←
SendServerKeyExchange (src/internal.c:22700) ←
wolfSSL_accept (src/ssl.c:9698) ←
server_test (examples/server/server.c:2077) ←
main (examples/server/server.c:2377) ←
```

If this is accurate, a simple solution might consist in zero-initializing variable `mp_int r_lcl` declared at `wolfcrypt/src/ecc.c:4444`.

There are possibly similar kinds of minor issues that I could report, but I don't know if they are interesting to you, since they may concern the specific implementation of the test server, with a specific set of macro definitions, and not the library itself (I'm using an interprocedural analysis instantiated on the code of the server), and there may be some false positives.

However, if you do find them useful, I can keep looking and see if I find others.
Also, these issues seem unlikely to be security-related, but if there are concerns, I can use a different channel to report them.

dgarske, 2019-07-23:
Hi maroneze,

Thanks for the report. I'll take a look at this. Yes please continue to send these types of reports. You can also direct them to our ZenDesk system using support@wolfssl.com.

Thanks,
David Garske, wolfSSL

kaleb-himes, 2019-07-23:
@maroneze 

> There are possibly similar kinds of minor issues that I could report, but I don't know if they are interesting to you

We NEVER discourage anyone poking around our code, we want to offer the most secure, best tested implementation available! Please send any/all reports, we gladly welcome any reports you are willing to send. Even if they don't pan out it gives us an opportunity to self-reflect and can lead to other positive improvements.

> However, if you do find them useful, I can keep looking and see if I find others.
> Also, these issues seem unlikely to be security-related, but if there are concerns, I can use a different channel to report them.

You can report via any channel but like David mentioned contacting us through zendesk via the support channel offers the best response times and tracking.

Warm Regards,

KH

closed 2019-09-20 by ejohnstown