wolfSSL/wolfssl #2343: VPATH builds seem broken

iorbitearth, 2019-07-11:
I noticed some issues with tests via `make check` when building outside the source tree (necessary as I'm compiling for multiple architectures). Traced similar issue back to building a level up from the source directory (see below). Is this a known issue? From my understanding VPATH build should work in any directory and this implies something is subtly wrong.

(version v4.0.0-stable)

Works:
```
$ ./autogen.sh
$ mkdir build_amd64 && cd build_amd64
$ ../configure
$ make
$ make check
$ echo $?
0
```

Fails:
```
$ autogen.sh
$ mkdir ../build_amd64 && cd ../build_amd64
$ ../wolfssl/configure
$ make
$ make check VERBOSE=1

<-- snip -->

PASS: scripts/openssl.test
PASS: scripts/external.test
PASS: scripts/google.test
FAIL: testsuite/testsuite.test
FAIL: tests/unit.test
PASS: scripts/resume.test
=====================================
   wolfssl 4.0.0: ./test-suite.log
=====================================

# TOTAL: 6
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  2
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: testsuite/testsuite
=========================

wolfSSL error: wolf root not found
FAIL testsuite/testsuite.test (exit status: 1)

FAIL: tests/unit
================

starting unit tests...
wolfSSL error: wolf root not found
FAIL tests/unit.test (exit status: 1)

============================================================================
Testsuite summary for wolfssl 4.0.0
============================================================================
# TOTAL: 6
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  2
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
```

kaleb-himes, 2019-07-11:
Hi @iorbitearth,

Thank you for your notes, the external directory is not support as we have no way of searching down in the file system without searching essentially your entire PC until wolf root is located and even then we wouldn't know if it was the right wolf directory. See we have certain test files (certificates) in certain locations and we have to determine where wolf-root is before the tests can execute so we check ../ if it's not root we go up again ../ and continue a certain number of times till we locate the root directory or we fail with the error ```wolf root not found```. This is an easy search.

In your Failing case if you check the log output you will see that it is the search for the root directory that is failing.

```
kalebs-MBP:build kalebhimes$ ./testsuite/testsuite.test 
wolfSSL error: wolf root not found
```

Searching the other direction (recursively down) is infinitely more complex and might end up locating the wrong root directory.

We will continue to support ```make check``` and all associated tests with the internal build like you noted in your "Works:" but if you are building outside the wolfSSL directory somewhere please expect both the unit.test and testsuite.test to fail with ```wolfSSL error: wolf root not found```.

Warm Regards,

KH

kaleb-himes, 2019-07-11:
@iorbitearth,

Is it a necessity for your use-case to have the external to wolf root case work for some reason? Can you tell us a bit about what you are attempting to do and why it would be necessary to build outside the wolfSSL directory?

Warm Regards,

KH

eblossom, 2019-07-12:
What the other person wants is standard when using configure and automake.
See the preset output vars $(srcdir) and $(top_srcdir) if you need to locate the source directory.
https://www.gnu.org/software/autoconf/manual/autoconf-2.67/html_node/Preset-Output-Variables.html

kaleb-himes, 2019-07-12:
@eblossom,

Thank you so much for your notes, we will look into the environment variables you noted!

Cheers,

KH

iorbitearth, 2019-07-15:
Thanks for taking a look.

This is the simplest example I could come up with to highlight the general issue. I believe it's related to `make check` failure when building/testing from a build directory under `wolfssl` root (with specific configuration option).

I will file a new issue to reproduce that more specific failure and link it to this open issue.

ejohnstown, 2019-07-16:
I believe that the actual VPATH build should work, but the "make check" is going to fail. We have some hardcoded paths in our test tools. I have a proof of concept for the example server where it is getting the srcdir though a CPPFLAG set by Automake.

You can see my PoC at https://github.com/ejohnstown/wolfssl/tree/vpath

iorbitearth, 2019-07-16:
You are right and a fix would be great. When building for various targets with various configuration options our team is not comfortable skipping the `make check` step.

PR #3725 (2021-02-08): build: fix `make distcheck`
Fixes #2343

`unit.test` now is generated pointing to unit test build directory.

Writing files in the dist is not allowed so during `distcheck` and out-of-tree build,
files are written to build directory.

Other tests handle referencing certs in source directory.
closed 2021-02-08 by toddouska