wolfSSL/wolfssl #2344: MSVC BUILD WOLFSSL_AESNI FOR X86 ERROR

zero-rp, 2019-07-12:
_mm_insert_epi64

zero-rp, 2019-07-12:
It seems that the encrypted result of CBC mode is not correct either.

ejohnstown, 2019-07-16:
Are you doing this as a 64-bit build or a 32-bit build?

ejohnstown, 2019-07-16:
Also, which commit from GitHub are you using and can you tell me how you are configuring the build?

zero-rp, 2019-07-19:
64-bit
vs2019
BUILDING_WOLFSSL
OPENSSL_ALL
OPENSSL_EXTRA

WOLFSSL_CUSTOM_CURVES
WC_NO_HARDEN
WOLFSSL_ALLOW_TLSV10
WOLFSSL_TLS13
HAVE_SUPPORTED_CURVES
WC_RSA_BLINDING
WOLFSSL_RIPEMD
NO_PSK
HAVE_EXTENDED_MASTER
WOLFSSL_SNIFFER
HAVE_TLS_EXTENSIONS
HAVE_SECURE_RENEGOTIATION
WOLFSSL_KEY_GEN
HAVE_COMP_KEY
WOLFSSL_DES_ECB
HAVE_FFDHE_2048
WC_RSA_PSS
HAVE_LIBZ
HAVE_PKCS7

HAVE_HKDF
HAVE_X963_KDF

HAVE_ECC
HAVE_ECC_ENCRYPT
HAVE_ECC_KOBLITZ

WOLFSSL_AES_CFB
WOLFSSL_AES_DIRECT
HAVE_AES_ECB
HAVE_AES_KEYWRAP
HAVE_AESGCM
HAVE_AESCCM

WOLFSSL_SHA384
WOLFSSL_SHA512
WOLFSSL_AESNI

build aes_asm.asm is 
ml64 /Fo $(IntDir)%(fileName).obj -c -coff -nologo -Zp4 -Cp -Cx /Zi /Zd /W3 %(FullPath)

zero-rp, 2019-07-19:
Are there any configuration errors?

dgarske, 2019-07-23:
Hi zero-rp,

We recently did some testing/improvements around the wolfCrypt benchmark Visual Studio project. In doing so we tested AESNI and Single Precision speedups on x64.

The build settings used are here:
https://github.com/wolfSSL/wolfssl/blob/master/IDE/WIN/user_settings.h#L51

The PR with these changes is here:
https://github.com/wolfSSL/wolfssl/pull/2361

Make sure your project has the pre-processor macro `WOLFSSL_USER_SETTINGS` defined.

Let us know if you are still seeing any build or run-time issues. 

Thanks,
David Garske, wolfSSL

zero-rp, 2019-07-28:
#ifndef _WIN_USER_SETTINGS_H_
#define _WIN_USER_SETTINGS_H_

#define WC_RSA_BLINDING

#define OPENSSL_ALL
#define OPENSSL_EXTRA

#define WOLFSSL_RIPEMD
#define NO_PSK
#define HAVE_EXTENDED_MASTER
#define HAVE_TLS_EXTENSIONS
#define WOLFSSL_ALLOW_TLSV10
#define WOLFSSL_TLS13

#define HAVE_AESGCM
#define HAVE_AES_ECB
#define HAVE_AES_KEYWRAP
#define HAVE_AESGCM
#define HAVE_AESCCM
#define WOLFSSL_AES_CFB
#define WOLFSSL_AES_DIRECT

#define WOLFSSL_SHA384
#define WOLFSSL_SHA512

/* Optional Performance Speedups */
#if 1
    /* AESNI on x64 */
#ifdef _WIN64
#define HAVE_INTEL_RDSEED
#define WOLFSSL_AESNI
#endif

/* Single Precision Support for RSA/DH 1024/2048/3072 and ECC P-256 */
#define WOLFSSL_SP
#define WOLFSSL_HAVE_SP_ECC
#define WOLFSSL_HAVE_SP_DH
#define WOLFSSL_HAVE_SP_RSA
#endif

#define HAVE_ECC
#define HAVE_ECC_ENCRYPT
#define HAVE_ECC_KOBLITZ
#define ECC_SHAMIR
#define ECC_TIMING_RESISTANT
#define WOLFSSL_CUSTOM_CURVES
#define HAVE_SUPPORTED_CURVES
#define HAVE_SECURE_RENEGOTIATION

#define WC_NO_HARDEN

#define WOLFSSL_KEY_GEN
#define HAVE_COMP_KEY

#define WOLFSSL_DES_ECB
#define HAVE_FFDHE_2048
#define WC_RSA_PSS
#define HAVE_LIBZ
#define HAVE_PKCS7

#define HAVE_HKDF
#define HAVE_X963_KDF

#endif /* _WIN_USER_SETTINGS_H_ */

zero-rp, 2019-07-28:
Aes aes;
    char a[16] = { 0 };
    char b[16] = { 0 };
    char c[16] = { 0 };
    wc_AesInit(&aes, NULL, 0);
    wc_AesSetKey(&aes, (const byte*)a, 16, NULL, AES_ENCRYPTION);
    wc_AesCbcEncrypt(&aes, (unsigned char*)b, (unsigned char*)c, (word32)16);

The result is incorrect after using WOLFSSL_AESNI

zero-rp, 2019-07-28:
@dgarske

zero-rp, 2019-07-28:
Master branch code used

dgarske, 2019-08-08:
Hi @zero-rp,

I suspect your wolfSSL library build settings are not matching your application.

Make sure your application code includes `#include <wolfssl/wolfcrypt/settings.h>` before any other wolfSSL headers. Also please make sure your application defines the pre-processor macro `WOLFSSL_USER_SETTINGS` and adds include paths for wolfssl-root and your user_settings.h file.

Thanks,
David Garske, wolfSSL

zero-rp, 2019-08-08:
I checked it carefully, because I didn't initialize the IV vector.When WOLFSSL_AESNI is used or WOLFSSL_AESNI is not used without IV vector, the encryption results are inconsistent.

dgarske, 2019-08-09:
Hi @zero-rp,

I can reproduce in Linux as well. Assigning this over to our AESNI expert Sean.

```c
Aes aes0;
char a[16] = { 0 };
char b[16] = { 0 };
char c[16] = { 0 };
wc_AesInit(&aes0, NULL, 0);
wc_AesSetKey(&aes0, (const byte*)a, 16, NULL, AES_ENCRYPTION);
ret = wc_AesCbcEncrypt(&aes0, (unsigned char*)b, (unsigned char*)c, (word32)16);
printf("AES 0 Test: Ret %d\n", ret);
WOLFSSL_BUFFER((byte*)b, 16);
wc_AesFree(&aes0);
```

```
./configure --enable-debug && make
./wolfcrypt/test/testwolfcrypt
...
AES 0 Test: Ret 0
	66 e9 4b d4 ef 8a 2c 3b 88 4c fa 59 ca 34 2b 2e | f.K...,;.L.Y.4+.

./configure --enable-aesni --enable-debug && make
./wolfcrypt/test/testwolfcrypt
...
AES 0 Test: Ret 0
	e5 a5 48 5f d2 f1 75 98 b8 02 b5 56 4a ae 2b 76 | ..H_..u....VJ.+v
```

Thanks,
David Garske, wolfSSL

zero-rp, 2019-08-09:
thinks

PR #2401 (2019-08-09): Set the IV to zero when NULL IV and using AESNI
Fixes #2344
closed 2019-08-09 by zero-rp