wolfSSL/wolfssl #2345: Please tell me how to achieve this.

zero-rp, 2019-07-12:
func aesGcmDecrypt(key,nonce,aad,data []byte)([]byte){
	block, err := aes.NewCipher(key)
	if err != nil {
		panic(err.Error())
		return nil
	}
	aesgcm, err := cipher.NewGCM(block)
	if err != nil {
		panic(err.Error())
		return nil
	}

	plain, err := aesgcm.Open(nil, nonce, data, aad)
	if err != nil {
		panic(err.Error())
		return nil
	}
	return plain
}

zero-rp, 2019-07-12:
The wc_AesGcmDecrypt function can be implemented, but the implementation of this function has errors.
AuhTag needs to pass null values, but the null values are exceptionally handled internally

ejohnstown, 2019-07-12:
I'm afraid I don't understand what you are trying to do. The AES-GCM specification doesn't allow for a 0 bit tag output. The purpose of the Galois Field multiply is to generate the tag. Otherwise use AES-CTR. Can you help me understand your use case?

zero-rp, 2019-07-12:
This is the go code translated above. This is a private protocol implemented by a Chinese company based on TLS1.3. The document is in Chinese. I don't know if you can understand it or not.

zero-rp, 2019-07-12:
[https://mp.weixin.qq.com/s?__biz=MzAwNDY1ODY2OQ==&mid=2649286266&idx=1&sn=f5d049033e251cccc22e163532355ddf&scene=0&key=b28b03434249256b2a5d4fdf323a185a798eaf972317ca3a47ef060d35c5cd8a4ae35715466d5bb5a558e424d20bef6c&ascene=0&uin=Mjc3OTU3Nzk1&devicetype=iMac+MacBookPro10%2C1+OSX+OSX+10.10.5+build%2814F1713%29&version=11020201&pass_ticket=8lpzOjRJO3IS%2BmKcvsqRN%2FlzlWyR2q2fmKv15GKO2PPYAKDGPXDhyfntueC4bIod](url)

zero-rp, 2019-07-12:
https://mp.weixin.qq.com/s?__biz=MzAwNDY1ODY2OQ==&mid=2649286266&idx=1&sn=f5d049033e251cccc22e163532355ddf&scene=0&key=b28b03434249256b2a5d4fdf323a185a798eaf972317ca3a47ef060d35c5cd8a4ae35715466d5bb5a558e424d20bef6c&ascene=0&uin=Mjc3OTU3Nzk1&devicetype=iMac+MacBookPro10%2C1+OSX+OSX+10.10.5+build%2814F1713%29&version=11020201&pass_ticket=8lpzOjRJO3IS%2BmKcvsqRN%2FlzlWyR2q2fmKv15GKO2PPYAKDGPXDhyfntueC4bIod

zero-rp, 2019-07-12:
In section 3.3.2 of the document, it is pointed out that the AES-GCM algorithm is used.The go code above is obtained by reverse engineering.His protocol has more than 1 billion users, so I didn't check the specification. If it's not a standard implementation, close the request.

zero-rp, 2019-07-12:
The go code above is executable

zero-rp, 2019-07-12:
wc_AesGcmDecrypt(&aes, out, buf, buf_len, nonce, nonce_len, NULL, 0, aad, add_len);This is the equivalent code.

zero-rp, 2019-07-12:
wc_AesGcmEncrypt Also

zero-rp, 2019-07-13:
https://golang.org/src/crypto/aes/aes_gcm.go

closed 2019-07-13 by zero-rp

ejohnstown, 2019-07-15:
I'm sorry for not replying. I read some of the document you linked to through Google Translate. I didn't see anything about requiring AES-GCM use without the tag. All mentions of AES-GCM stated it did encryption AND authentication. To ignore the tag is to ignore the authentication.

I have seen some cases where the authentication tag is appended to the end of the cipher text as a part of the output. To do that with wolfCrypt, you would have a wrapper function that receives the output buffer that is sizeof(input) + sizeof(tag) in length. Your wrapper function would call our wc_AesGcmEncrypt() and supply the output buffer as the output pointer and &output[sizeof(input)] as the tag pointer.