wolfSSL/wolfssl #2355: Port wolfssl to wio gps board

jim-king-2000, 2019-07-17:
I'm porting wolfssl to wio gps board("[http://wiki.seeedstudio.com/wio_gps_board/](http://wiki.seeedstudio.com/wio_gps_board/)"). The MCU is "ATSAMD21G18A" the network module is a GSM/GPRS model "Quectel MC20", and the development environment is Arduino.
Now I meet the error like this: "#error "Must define XGMTIME externally see porting guide"". I don't think the board can support the time function in the standard c run-time. How can I fix the error?

dgarske, 2019-07-17:
Hi Jim-King-2000,

If your board does not have a way to get time/RTC you can disable all time support using the `NO_ASN_TIME` build option. Note: this will disable all date/time checking for certificates and would allow an expired certificate via TLS.

Also wolfSSL does have its own implementation of time. Try defining `WOLFSSL_GMTIME`.

If you want to implement your own time function you can define something like the following:

```
#define USER_TICKS
extern unsigned long my_time(unsigned long* timer);
#define XTIME my_time
```

At a minimum I would recommend capturing the last firmware compile date/time into a variable for before certificate date validation using `__DATE__` and `__TIME__`. You can use our verify callback function set via `wolfSSL_CTX_set_verify` to define your own certificate date handling. See https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/test.h#L1717

Thanks,
David Garske, wolfSSL

jim-king-2000, 2019-07-18:
Hi David,

I build it successfully after reading your comments. However, it hangs somewhere. The log is like the following:

23:48:48.198 -> wolfSSL Leaving DoHandShakeMsgType(), return 0
23:48:48.198 -> wolfSSL Leaving DoHandShakeMsg(), return 0
23:48:48.198 -> Data received
23:48:48.198 -> 	16 03 03 13 53                                  | ....S
23:48:48.198 -> growing input buffer
23:48:48.198 -> 
23:48:48.721 -> Data received
23:48:48.721 -> 	0b 00 13 4f 00 13 4c 00 05 e4 30 82 05 e0 30 82 | ...O..L...0...0.
23:48:48.721 -> 	04 c8 a0 03 02 01 02 02 10 0f 70 19 fd f1 73 d0 | ..........p...s.
23:48:48.721 -> 	e6 df cf c9 31 ad 90 2a 26 30 0d 06 09 2a 86 48 | ....1..*&0...*.H
23:48:48.721 -> 	86 f7 0d 01 01 0b 05 00 30 46 13 06 03 55 04 53 | ........0F...U.S
23:48:48.721 -> 	65 72 76 65 72 20 43 41 20 31 42 31 0f 30 0d 06 | erver CA 1B1.0..
23:48:48.721 -> 	03 55 04 03 13 06 41 6d 61 7a 6f 6e 30 1e 17 0d | .U....Amazon0...
23:48:48.721 -> 	31 39 30 35 30 39 30 30 30 30 30 30 5a 17 0d 32 | 190509000000Z..2
23:48:48.721 -> 	30 30 36 30 39 31 32 30 30 30 30 5a 30 15 31 13 | 00609120000Z0.1.
...

The part of  the last "Data received" is quite large. But there is no more log. It hangs. Any clue to resolve this?

Thanks,
Jim Jin

dgarske, 2019-07-18:
Hi Jim-King-2000,

That's great you got things working that far. Do you have a debugger so you can break and see the call stack where it has hung? I suspect its a heap or stack memory issue. I recommend checking those settings and increasing both heap/stack to maximum possible and if works then reduce. Depending on the cipher suite TLS can take from 8-20KB stack and heap.

Thanks,
David Garske, wolfSSL

jim-king-2000, 2019-07-19:
Hi David,

Thanks for your quick reply and patience.

1. I don't have a debugger. I'm using Arduino IDE to download the HEX to the board by means of USB (virtual serial port actually). I can only watch the log.
2. My MCU is ATSAMD21G18A, which contains 256KB flash and 32KB RAM. However, I'm still not sure if it is enough for wolfSSL.
3. It seems that there is no option or setting for stack/heap in Arduino environment. I know I shouldn't use Arduino. But it is very appropriate for a quick prototype.

I'll update it here if I find something new.

Thanks,
Jim Jin

jim-king-2000, 2019-07-19:
Hi David,

The culprit is the recursion in the function "WOLFSSL_BUFFER" in "logging.c". It leads to stack overflow when dumping a large number of data. After changing it to loop, it goes farther.

I suggest that any recursion should be eliminated in wolfSSL since we can never predict how deep it could be.

There are still some run-time errors and I'm keeping investigating.

Thanks,
Jim Jin

jim-king-2000, 2019-07-19:
> The new error message is:
> Loading peer's cert chain
> 12:33:35.028 -> 	Put another cert into chain
> 12:33:35.028 -> wolfSSL Leaving ProcessPeerCerts, return -328

It seems that I have to study cert chain.

Thanks,
Jim Jin

jim-king-2000, 2019-07-19:
The error is raised here (function "ProcessPeerCerts" in internal.c):

                `if ((args->idx - args->begin) + OPAQUE24_LEN > totalSz) {
                    ERROR_OUT(BUFFER_ERROR, exit_ppc);
                }`

What does it mean? And how to make it right?

jim-king-2000, 2019-07-20:
The root cause is that the buffer of the serial port overruns and it leads to data loss. I resolved it by increasing the buffer.

jim-king-2000, 2019-07-21:
Hi David,

I port wolfSSL successfully. Thank you.

Regards,
Jim Jin

closed 2019-07-21 by jim-king-2000