wolfSSL/wolfssl #2368: Having problem enable TLS1.3 with LINUX_SGX

RyukerLiu, 2019-07-22:
I try to enable TLS1.3 with LINUX SGX, but encounter some problem.

After compare the makefile between configure enable-tls1.3. I try to add some flag and library.

`$(WOLFSSL_ROOT)/wolfcrypt/src/sha3.c\`
`$(WOLFSSL_ROOT)/src/tls13.c\`
`+Wolfssl_C_Flags := $(Flags_Just_For_C) $(Common_C_Cpp_Flags) $(Wolfssl_C_Extra_Flags) -DHAVE_FFDHE_2048 -DWOLFSSL_TLS13 -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -D_POSIX_THREADS -DHAVE_HKDF -DWC_RSA_PSS`

It can compile successfully. But when I try to run the sgx example with TLS 1.3 method.
It will encounter illegal instruction (core dumped) when calling enc_wolfSSL_Init.

kaleb-himes, 2019-07-22:
Hi RyukerLiu,

Thanks for reaching out to wolfSSL support. Can you tell us what it is you are working on and what is driving the effort?

The defines to add TLS 1.3 support should be:

```
#define WOLFSSL_TLS13
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES
#define HAVE_ECC
#define HAVE_HKDF
#define HAVE_FFDHE_8192 // or other FFDHE option, at least one is required.
#define WC_RSA_PSS
```

Once these have been added to the static lib Makefile you need to also add these same settings to the SGX application Makefile so the library and application both have common configuration.

Warm Regards,

KH

RyukerLiu, 2019-07-24:
Hi kaleb-himes,

Thank you. It's worked.

I write some CLI in SGX. And want to hold TLS server to be the bridge to the web and CLI tool.

So I'm working on it.

closed 2019-07-26 by RyukerLiu